hpwit/FASTLEDVIRTUALDRIVER #2: More Data Line With Less Wires From Esp32

ScobbyDoo, 2020-12-15:
ignore this points at initial stage
* Signals' level shifter from esp32 to 74hc595
* Huge Current for led
* Voltage drop

[image]

as per you worked around 74hc595 ,is it possible to drive this kind of setup(256 data line)?

hpwit, 2021-02-12:
Hello
The idea is good but the issue is the required speed to do so. the maximum output speed in I2s is 20Mhz
to push the 8:1 I am already @19,2Mhz.
regards

midikidi, 2021-11-28:
Yves, you know better that me but i found this
For more information, see ESP32 Technical Reference Manual > I2S Controller (I2S) > LCD Mode [ PDF ]. For high accuracy clock applications, use the APLL_CLK clock source, which has the frequency range of 16 ~ 128 MHz. You can enable the APLL_CLK clock source by setting i2s_config_t::use_apll to TRUE.

hpwit, 2021-11-29:
hello
That is the clock I am using already. The issue is not the speed but the fact that at that speed sometimes bits are missed when transmitting. for audio it's not issue but for digital like for the leds then it's a problem. It has also been noticed by other developers. I have at that time studied it

closed 2021-12-20 by hpwit