hpwit/FASTLEDVIRTUALDRIVER #3: Works with other platform ?

Kamal-Sonani, 2020-12-19:
Esp8266 

Or avr platform ? 
(Atmega8 or 328)

As your sketch used 2 core esp32 is it compatible with single core processor? 

On avr platform communication through Enc28j60 Ethernet

hpwit, 2020-12-19:
The virtual driver make use of the i2s of the esp32. So I don’t know for esp8266. I need to look at the i2s of the esp8266. but for sure it could work using the spi. How many strips do you want

ScobbyDoo, 2021-01-08:
@hpwit did you tried virtual output in esp8266 yet?
I found port masking logic in your code for esp32 (16 parallel output)
which way is quicker for multiple output (as compered direct gpio parallel or virtual driver)

I want to try with artnet OR E1.31 & also with sd card

hpwit, 2021-01-08:
hello
the virtual driver is made is you want a lot of parallels pins. how many pins do you want ?
I haven't looked at the esp8266 yet
Yves

ScobbyDoo, 2021-01-10:
I'm trying to drive 8 output with 8 gpio without any other process
I'm getting this type of output on each strip 
**Process 1**
first process data~ for pin1, after that for pin2 & so on
pin1 -----
pin2 _____-----       
pin3 __________-----      
pin4 _______________-----
pin5 ____________________-----       
pin6 _________________________-----  
pin7 ____________________________----- 
pin8 _________________________________-----     

instead of  (Don't know how to drive parallelly as port)
**Process2** by doing this FPS will be increased
pin1 -----
pin2 -----     
pin3 -----     
pin4 -----     
pin5 -----     
pin6 -----     
pin7 -----     
pin8 -----               

i would like to drive 16 output with esp8266 with 4gpio as you did in ESP32 using shift registers 
no matter what FPS i get

ScobbyDoo, 2021-06-28:
i want to drive 16 parallel with esp8266 
nearly 100 to 150 led on each data line

i want to split leds instead of in serial feed

hpwit, 2021-06-28:
Hello 
To create the 16 parallel outputs I will need to check for the esp8266. I can do 8 // outputs.

ScobbyDoo, 2021-06-29:
I tried with i2s but not getting right way to do it , compared with your esp32's logic flow.
direct spi used in SD card (logging data),so I'am trying on Soft SPI
Yes , I stuck at 8266

ScobbyDoo, 2021-07-05:
can you do it? , I want it desperately 
on payable scale

hpwit, 2021-07-05:
I will look at it. 
how many // outputs do you need ?

ScobbyDoo, 2021-07-06:
8 

& 16 if possible

ScobbyDoo, 2021-07-11:
@hpwit what do yo think?

hpwit, 2021-07-11:
I will look at it this week in more detailed. I have found an esp8266 in my drawer ;)

ScobbyDoo, 2021-07-23:
What will be initial setup to drive high speed & time critical virtual pins?

I mean which logic should i try?
Using i2s or simply SPI or using spi & DMA?

hpwit, 2021-07-23:
I use i2s and dma.

ScobbyDoo, 2021-09-01:
@hpwit i found some esp8266 in last changes , did it works?

ScobbyDoo, 2021-09-15:
I checked repo , found some esp8266's revision.
It it working with 595?

hpwit, 2021-09-16:
Hello
Sorry I was busy with other stuff. NO I haven't looked at it yet
i will soon

ScobbyDoo, 2021-09-16:
@hpwit & @Kamal-Sonani 

  Is their any option for paid service? I want this very desperately 😅 it will be easy for me , working on this from long time but no luck.

Any help would be great.
Thanks

hpwit, 2021-09-18:
Hello
I will have a look at it. Why can’t you change to esp32. But I will try to make it work

closed 2021-12-20 by hpwit