hpwit/FASTLEDVIRTUALDRIVER #4: clockless_i2s_virtual_esp32.h missing?

Munger, 2021-02-06:
I'm trying to build the FastLEDVirtualDriver examples and clockless_i2s_virtual_esp32.h is not found. I have built a 40 pin board and am not entirely sure which eaxample I should be looking at and how to configure the #defines for this board. Please advise.

hpwit, 2021-02-06:
Hello you need to build this 
https://github.com/hpwit/FASTLEDVIRTUALDRIVER/blob/master/examples/VirtualdriverExamples/VirtualDrivers8pins/VirtualDrivers8pins.ino
I need to remove the 5 pin version 
This version would make you drive 8 strips per pins.

Munger, 2021-02-06:
Will this work with the 40 pin PCB? I already had those made

hpwit, 2021-02-07:
try this 
https://github.com/hpwit/FASTLEDVIRTUALDRIVER/tree/cb58824406d95813412ec193a6668fd2759788f4
this is the code with the 5:1 version

Munger, 2021-02-08:
That sort of worked, thanks. I changed the example code to just display red pixels but It's showing alternate Yellow/Red/Blue, so I'm not sure what's going on there. Looks more like a code bug than a hardware problem so I can delve a little deeper into that.

Out of curiosity, how do we specify what chipset the LEDs use, or is that not relevant?

hpwit, 2021-02-09:
Hello
can I see the code
Di you take the entire library or just the file ?
It's been a long time I haven't looked at the 5:1 version. which pub did you build ? the 5:1 or the 8:1 ?
regards yves

hpwit, 2021-02-09:
for the moment the virtual driver works for ws281x leds only it's not parametrable for other chipset yet
I have wrote the parallel output for apa102 I could write the virtual driver for it too. But I have never seen any need for it yet but I you want I can write it

Munger, 2021-02-09:
I took the entire library  (https://github.com/hpwit/FASTLEDVIRTUALDRIVER/tree/cb58824406d95813412ec193a6668fd2759788f4 ) and made a trivial change to the example sketch as below

I built this board - https://github.com/hpwit/FASTLEDVIRTUALDRIVER/blob/master/examples/VirtualdriverExamples/VirtualDrivers5pins/Schematic_40-pins-2LS_40-pins-2LS_20190811155444.png

BTW, I'm currently designing a 64-pin board with the ESP32 built-in. Have you tried daisychaining the shift registers and driving them all off of three pins?

-----
#define ESP32_VIRTUAL_DRIVER true //To enable virtual pin driver for esp32
#define NBIS2SERIALPINS 8

/*
This needs to be declared before #include "FastLed.h"
NBIS2SERIALPINS represent the number of pins used by the esp value, it ranges from 1->20
for each pin you would be able to drive 5 virtuals pins
i.e for
#define NBIS2SERIALPINS 8
You will be able to drive up to 8*5=40 pins
If you want to have a number of strips which is not a multiple of 5 no issue
if you want to drive 21 pins you would need #define NBIS2SERIALPINS 5
if you want to drive 17 pins you would need #define NBIS2SERIALPINS 4
*/
#include <Arduino.h>
#include "FastLED.h"`
#define LATCH_PIN 12
#define CLOCK_PIN 27 //for a reason I don't know, the CLOCK_PIN needs to be >=16

#define NUM_LEDS_PER_STRIP 256 //the length of your strip if you have different strips size put here the longuest strip
#define NUM_STRIPS 40
#define NUM_LEDS NUM_LEDS_PER_STRIP * NUM_STRIPS
CRGB leds[NUM_LEDS];

int Pins[NBIS2SERIALPINS]={13,14,26,25,33,32,15,18};

void setup() {
Serial.begin(115200);
/*
* Define the esp pins you are using
* pin 13 will drive strip 1->5
* pin 14 will drive strip 6->10
* pin 26 will drive strip 11->15
* pin 25 will drive strip 16->20
* pin 33 will drive stip 21->25
* pin 32 will drive strip 26->30
* pin 15 will drive strip 31->35
* pin 18 will drive strip 36->40
*/

FastLED.addLeds<VIRTUAL_DRIVER,Pins,CLOCK_PIN, LATCH_PIN>(leds,NUM_LEDS_PER_STRIP);
FastLED.setBrightness(64);

}
int start=0;
void loop() {
    //fill_solid(leds, NUM_LEDS, CRGB(15,15,15));
    fill_solid(leds, NUM_LEDS, CRGB(0,0,0));
    /*
    * this code will create a snake on each strip where the length is the strips number
    */
    for(int i=0;i<NUM_STRIPS;i++)
    {
        int offset=i*NUM_LEDS_PER_STRIP;   //this is the offest of the strip number i
        for(int k=0;k<i+1;k++) {
            leds[(start+k)%NUM_LEDS_PER_STRIP+offset]=CRGB::Red; //CHSV(i*255/NUM_STRIPS,255,255);
        }

    }
    long     lastHandle = __clock_cycles();
    FastLED.show();
    long   lasthandle2=__clock_cycles();
    Serial.printf("FPS fastled: %f \n", (float) 240000000L/(lasthandle2 - lastHandle));
    start++;
    delay(30);

}

hpwit, 2021-02-09:
Hello
I will check with my 5:1 boards.I still have a couple of them. 
The ws2812 is at 800khz to send the signal. 
To get the one and zero wee do send 110 (for 1) and 100 for zeros this 3 bits need to be send on 1250ns(800khz) for all three hence  each bit will be sent to a frequency of 3x800khz or 2.4Mhz. 

Now to send 1 bit over the shift register all the data needs to be in the shift register before latch when the data is pushed to the output of the shift register. Hence the latch frequency is 2.4 MHz. But before that i need to push 5 (or 8 ) bits in the shift register. 
We have to push the bits in the shift register as fast as we can.
So if we want to push n bits to the shift register they need to be pushed at n x 2.4Mhz speed (at least). 
 But using i2s above 20mhz some bits are missing and not pushed (i2s is made for sound hence not really bad) hence n=20/2.4 MHz =8 
That is why I cannot daisy chain the shift registers. The esp32 will not be able to go up to the required frenquécies. 
I hope I was clear

hpwit, 2021-02-09:
Hello
I have tried on my 40 pins board 5:1 and it works fine. Could you make a pic of your board ?  Front and back ?

Munger, 2021-02-09:
I see. Thanks for the explanation
[image]

One more question. While fiddling with routing on my design for a 64 pin board, it occurred to me that the clock outputs from the 74HC245 are basically identical as the inputs are bridged (same for the latch outputs. I come from a programming background and am a complete novice at electronics. Wouldn’t it be possible to simply use two pairs of pins on the 245s and wire the clock and latch inputs of the shift registers in parallel?

[image]

hpwit, 2021-02-12:
I have looked at the gerber file
I don't see anything wrong. where did you buy your ICs ?
if you want I can send you the board I am using to test (I am not using it)
the 245 are there to help the for two things 
Protecting the Mcu and getting a slight delay that makes the clock goes round :) especially for the 8:1

Munger, 2021-02-12:
The chips came from Mouser, so I doubt there is a problem there. Thanks for the offer , but I’m m sure I’ll figure it out when I get my  debugger working. 

I understand what the 245s are doing, but that isn’t what I’m asking. Because you have linked the photo to inputs, the outputs are also logically identical. Is there a good reason for doing this? Why not just use a single output wire from the 245 and can next it to all the clock ping of the shift registers? Likewise for the latch lines. Am I missing something about the way the 245 works? Now that I think about it, wouldn’t a transistor provide the same functionality?

like I said, I’m very new to hardware design and am probably missing a few subtleties.

hpwit, 2021-02-12:
Hello
I   have asked the question about the Ics because it happened to me.
do you  have the error on all the pins ?
if you have the color changing that means that there is a timing issue between the latch and the clock.
It will be interesting to have an oscilloscope to check the outputs.
When using a 245 you cannot let an input not plugged if you do that you risk to to have issue on all outputs. Indeed I could have used only one 245 and then plugged all but two to ground  one to clock and on t to latch. 
Using a transistor you come to a speed issue. you would need a really fast mos transistor to do the equivalent here we have digital signal @couple Mhz. plus an IC as a really good 'inifinite' entry impedance. at a very low cost.
If you want to reduce size you can use smd components.
if you can send a video of what's happening. I am trying to get my head around what happen
Can I ask you to check that all IC receive the 5V (I had a similar issue on a previous board=> hence the version 1.0)

Munger, 2021-02-12:
Here’s a video as requested. I failed to mention that I’m still trying to set all pixels in the snake to CRGB::Red. I’m trying to figure out if I can send you my compiled firmware.

hpwit, 2021-02-12:
By any bad luck aren’t your leds rgbw ?

Munger, 2021-02-12:
Yes. but when I use them with XLights/WLED etc I choose WS8212B with no issues. This is why I asked if it was possible to specify the exact chipset with the virtual driver.

Munger, 2021-02-12:
But now it's pretty obvious what the issue is :D

hpwit, 2021-02-12:
That is super strange it’s like it’s skipping bytes. It’s like it was trying to display rgbw instead of rgb. Can you zip your fastLED library please and mail it to me ? BAZIN.Yves gmail com

Munger, 2021-02-12:
It makes perfect sense, The leds are expecting 4 bytes per pixel and only receiving 3. If I can locate the function that actually outputes the bytes I can fudge in a 0 byte for the white value and I fully expect it to work.

Munger, 2021-02-12:
WLED allows us to specify RGBW with the WS2812 type strips. I had foolishly assumed that FastLED supported RGBW. I see that Dan Garcia had been working on that, but was sadly taken from us.

hpwit, 2021-02-12:
Ok if they are rgbw leds I know what to do ;) I did it for the 8:1

hpwit, 2021-02-12:
But I will need a bit of time to reprogram it on this version of the driver

hpwit, 2021-02-12:
But you can always says the length being 150x4/3=200

hpwit, 2021-02-12:
change 
#define NUM_LEDS_PER_STRIP 256
add this below
 uint8_t *dleds

int setup function add
dleds=(uint8_t *)leds;

change 
for(int i=0;i<NUM_STRIPS;i++)
{
int offset=I*150*4; //this is the offest of the strip number i
for(int k=0;k<i+1;k++) {
dleds[((start+k)%150)*4+offset+1]=255;
}

Munger, 2021-02-12:
Thanks, that causes it to use the whole strip, but the colours will still not be correct with an RGBW strip. Whereever the code is that actually outputs the data to the leds is, it still assumes 3 bytes per pixel which is what causes the colours to cycle . I think there will be far too much work making FastLED compatible with RGBW strips. Ultimately I will be using this with RGB pixel strings. I just don't own any of those yet.

Munger, 2021-02-12:
The easiest thing would be to keep everything as RGB and simply convert to RGBW at the output phase. This is basically how under-colour removal works with CMYK printing.

https://www.reddit.com/r/FastLED/comments/ht0jei/how_to_rgb_rgbw_conversion_with_gamma_and_color/

hpwit, 2021-02-12:
Do you at least have the same color or does the color changes.

hpwit, 2021-02-12:
I could do a quick change so that the program outputs 32 bits instead of 24. I will make a change tomorrow

Munger, 2021-02-12:
When I specify all red pixels, it outputs Red-Blue-Green. The red value for the second pixel is being eaten by the strip as the white value for the first pixel. This effectively means that the second pixel is seen by the strip as 0,0,255,0, so it displays as blue.

Can you tell me where the value are output?

hpwit, 2021-02-13:
you cannot use the CRGB obect fro the Fastled lib yet
but normally if you use this 
for(int i=0;i<NUM_STRIPS;i++)
{
int offset=ix150x4; //this is the offest of the strip number i
for(int k=0;k<i+1;k++) {
dleds[((start+k)%150)x4+offset+1]=255;
}
instead of the other code in the loop the led should turn red only
 can you check ?

hpwit, 2021-02-13:
ok it's done (first version) reupload the master version not the link I gave you.
add 
#define NUM_COLOR_CHANNELS 4
before #include "FastLED.h"

and use your FORMER CODE I mean the one with #define NUM_LEDS_PER_STRIP 150 ...
and 
  for(int i=0;i<NUM_STRIPS;i++)
    {
        int offset=i*NUM_LEDS_PER_STRIP;   //this is the offest of the strip number i
        for(int k=0;k<i+1;k++)
        {
            leds[(start+k)%NUM_LEDS_PER_STRIP+offset]=CRGB::Red;
        }

    }

let me know

Yves

Munger, 2021-02-13:
#define ESP32_VIRTUAL_DRIVER true //To enable virtual pin driver for esp32
#define NBIS2SERIALPINS 8
#define NUM_COLOR_CHANNELS 4

/*
This needs to be declared before #include "FastLed.h"
NBIS2SERIALPINS represent the number of pins used by the esp value, it ranges from 1->20
for each pin you would be able to drive 5 virtuals pins
i.e for
#define NBIS2SERIALPINS 8
You will be able to drive up to 8*5=40 pins
If you want to have a number of strips which is not a multiple of 5 no issue
if you want to drive 21 pins you would need #define NBIS2SERIALPINS 5
if you want to drive 17 pins you would need #define NBIS2SERIALPINS 4
*/
#include <Arduino.h>
#include "FastLED.h"`
#define LATCH_PIN 12
#define CLOCK_PIN 27 //for a reason I don't know, the CLOCK_PIN needs to be >=16

#define NUM_LEDS_PER_STRIP 150 //the length of your strip if you have different strips size put here the longuest strip
#define NUM_STRIPS 40
#define NUM_LEDS NUM_LEDS_PER_STRIP * NUM_STRIPS

//CRGB leds[NUM_LEDS];
//uint8_t *dleds = (uint8_t *)leds;

uint8_t dleds[ NUM_LEDS_PER_STRIP * NUM_STRIPS * NUM_COLOR_CHANNELS ];
CRGB* leds = (CRGB*) dleds;

int Pins[NBIS2SERIALPINS]={13,14,26,25,33,32,15,18};

void setup() {
Serial.begin(9600);
/*
* Define the esp pins you are using
* pin 13 will drive strip 1->5
* pin 14 will drive strip 6->10
* pin 26 will drive strip 11->15
* pin 25 will drive strip 16->20
* pin 33 will drive stip 21->25
* pin 32 will drive strip 26->30
* pin 15 will drive strip 31->35
* pin 18 will drive strip 36->40
*/

FastLED.addLeds<VIRTUAL_DRIVER,Pins,CLOCK_PIN, LATCH_PIN>(leds,NUM_LEDS_PER_STRIP);
FastLED.setBrightness(64);

}
int start=0;
void loop() {
    fill_solid(leds, NUM_LEDS, CRGB(15,15,15));
    //fill_solid(leds, NUM_LEDS, CRGB(0,0,0));
    /*
    * this code will create a snake on each strip where the length is the strips number
    */
#if NUM_COLOR_CHANNELS==4
    for(int i=0;i<NUM_STRIPS;i++) {
        int offset=i*NUM_LEDS_PER_STRIP*NUM_COLOR_CHANNELS; //this is the offest of the strip number i
        for(int k=0;k<i+1;k++) {
            dleds[((start+k)%NUM_LEDS_PER_STRIP)*NUM_COLOR_CHANNELS+offset+1]=255;
        }
    }
#else
    for(int i=0;i<NUM_STRIPS;i++){
        int offset=i*NUM_LEDS_PER_STRIP;   //this is the offest of the strip number i
        for(int k=0;k<i+1;k++) {
            leds[ (start+k)%NUM_LEDS_PER_STRIP+offset]=CRGB::Red; //CHSV(i*255/NUM_STRIPS,255,255);
        }
    }
#endif

    long     lastHandle = __clock_cycles();
    FastLED.show();
    long   lasthandle2=__clock_cycles();
   // Serial.printf("FPS fastled: %f \n", (float) 240000000L/(lasthandle2 - lastHandle));
    start++;
    delay(30);

}

hpwit, 2021-02-13:
No you should not have a specific code for num_color_channels is 4. 
always use the same code 
for(int i=0;i<NUM_STRIPS;i++){
int offset=iNUM_LEDS_PER_STRIP; //this is the offest of the strip number i
for(int k=0;k<i+1;k++) {
leds[ (start+k)%NUM_LEDS_PER_STRIP+offset]=CRGB::Red; //CHSV(i255/NUM_STRIPS,255,255);
}

Munger, 2021-02-13:
Ah! There we go! My bad, I misread your previous comment. It's working brilliantly.

hpwit, 2021-02-13:
Great !!!
At least it was not the board. 
 it’s not really full rgbw because it outputs always 0 for the white led. But like that it’s fully compatible with fastled. 
Tell me if you want full rgbw capabilities or I can transform rgb~>rgbw. 
inwould love to see when you have more panels linked Together.

Munger, 2021-02-13:
I don’t think I’ll ever need full RGBW. Ultimately this will be an E1.31 pixel controller for use with XLights.  That will be driving RGB pixel strings.

hpwit, 2021-02-13:
Oki let me know if you need more help 
Everything is fine on all 40 strips ?

ScobbyDoo, 2021-04-10:
1:- Can we drive DMX here?
Like each gpio will feed 1 universe.

2:- How to set timing for other custom led's protocol?

Munger, 2021-04-10:
> 1:- Can we drive DMX here?
> Like each gpio will feed 1 universe.
> 
> 2:- How to set timing for other custom led's protocol?
> 
> 3:- this example works with esp8266 ? Since esp8266 have i2s too that's why I'm asking. Only with 1 shift register

Yes. I did it using this - https://github.com/forkineye/ESPAsyncE131 and managed to drive 64 strings. 

However, I’m not convinced this is ideal as the driver renders an entire frame every time we call ShowLeds(). I feel this library is perfectly suited to large matrix displays rendering video, but if you are thinking about driving a pixel string display with something like XLights then updating the entire frame for each received packet could cause lag.

hpwit, 2021-04-11:
hello @ScobbyDoo 
1) Yes you can use DMX 
2) you cannot set a specific timing for custom led's protocol. but if you have a type in mind I will be more than happy to see if it is possible.
3) I am using the I2S to have parallel output because I need to create at least a latch signal. I don't know if you can do parallel I2S with esp8266 I will look at it

Do you have a specific project in mind ?

ScobbyDoo, 2021-04-11:
Specific information of project be like

Play E1.31 OR artnet streaming using Ethernet (SPI) on esp8266 & using extra 3 pin  esp got 8 virtual pins for pixel

Any other Board can't fulfill this requirement (prise wise Esp8266 is cheaper)

closed 2021-12-20 by hpwit