hpwit/FASTLEDVIRTUALDRIVER #5: c array not working

dallen98, 2021-02-21:
Hello Yves
I am trying to get the following sketch to work testing with a 16x16 neopixel panel, I have tested where the length is the strips number and that works accross 60 pins coming from the 74hc595 chips, i have solderd 3 wires to each of the 74hc595 to get the 8.1 output.
any suggestions much appreciated

//#define FASTLED_INTERNAL  //Elminates a compiler warning
#include <pgmspace.h>  // Needed to store stuff in Flash using PROGMEM

//disable psram
//#define ESP32_VIRTUAL_DRIVER true //To enable virtual pin driver for esp32
//#define ESP_VIRTUAL_DRIVER_8 1 //to use the 8:1 ratio
//#define FULL_DMA_BUFFER
//#define NBIS2SERIALPINS 12  //12 pins max, any more is unstable
#define NBIS2SERIALPINS 1
//#define NBIS2SERIALPINS 10     //number of esp32 pins you will use. the total number of strips available will be NBIS2SERIALPINS * 8 here 72 strips
//#define NUM_LEDS_PER_STRIP 900 //the length of your strip if you have different strips size put here the longuest strip
#define LED_WIDTH 16
#define LED_HEIGHT 16
#define NUM_LEDS LED_WIDTH*LED_HEIGHT
#define NUM_LEDS_PER_STRIP 256   //16 x 16 ws2812b panel
#define MULTIPLE_LEDSBUFFER //use the mode several leds buffers in that mode the number of NBIS2SERIALPINS has to be <=12
#define STATIC_COLOR_GRB 1
#define USE_FASTLED //te enable the needed types
#include "FastLED.h" //this is the 8.1 yves version from https://github.com/hpwit/FASTLEDVIRTUALDRIVER
#include "I2SClocklessVirtualLedDriver.h"
#define LATCH_PIN 13
#define CLOCK_PIN 27 //for a reason I don't know, the CLOCK_PIN needs to be >=16

#define NUM_STRIPS NBIS2SERIALPINS * 8
#define NUM_LEDS NUM_LEDS_PER_STRIP *NUM_STRIPS

//int Pins[15] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4, 17, 16, 21, 23, 19};
//int Pins[12] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4, 17, 16};  //12 mins max
//int Pins[10] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4};  //12 pins max
int Pins[1] = {12};  //12 pins max
/*
* Define the esp pins you are using // 900 leds per pin max
* pin 12 will drive strip 1->8
* pin 14 will drive strip 9->16
* pin 26 will drive strip 17->24
* pin 25 will drive strip 25->32
* pin 33 will drive stip  33->40  //40 x 900=36000
* pin 32 will drive strip 41->48
* pin 5  will drive strip 49->56
* pin 15 will drive strip 57->64
* pin 2  will drive strip 65->72  // 72 x 900=64800
* pin 4  will drive strip 73->80  // 80 pins x 900 leds=72000 leds 
*                                 // target 64800 (180 x 360= 64800)
* pin 17 will drive strip 81->88  //not required
* pin 16 will drive strip 89->96  //not required
*/

I2SClocklessVirtualLedDriver driver;
const unsigned long one[] PROGMEM =  //this is 16x16 pixels "1" in blue on red background
{
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf80305, 0x83365c, 0x355998, 0xa52744, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xfe0000, 0xc6182a, 0x2062a8, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x046ebd, 0x46518b, 0xbd1d31, 0xf70306, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xa92640, 0x265fa3, 0x0070c0, 0x0070c0, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x2f5c9d, 0xa02a47, 0x365898, 0x7e3961, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xeb080e, 0x3a5694, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x3a5694, 0xeb080e, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xeb080e, 0x3a5694, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x3a5694, 0xeb080e, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xeb080e, 0x3a5694, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xfa0204, 0xa22846, 0x674272, 0x893358, 0xf30508, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000
};
void setup()
{
    Serial.begin(115200);

    driver.initled(NULL, Pins, CLOCK_PIN, LATCH_PIN);
    driver.setBrightness(20);
}
int start = 0;
void loop()
{
    // memset(leds,0,NUM_LEDS*3);
    /*
    * this code will create a snake on each strip where the length is the strips number
    */
    for (int i = 0; i < NUM_STRIPS; i++)  {
    //for(int i = 0; i < NUM_LEDS; i++) {
      //driver.LEDS[i] = pgm_read_dword(&(one[i]));  // Read array from Flash
    //}
        //int offset = i * NUM_LEDS_PER_STRIP; //this is the offest of the strip number i
        for (int k = 0; k < (i+1); k++)
        {
//because the strip is split between several led buffers this is how you can set a color
            //driver.LEDS[k % NUM_LEDS_PER_STRIP + offset] = CHSV(k + start, 255, 255);  //pin count rainbow
            driver.LEDS[i % NUM_LEDS_PER_STRIP] = pgm_read_dword(&(one[i]));
          
            //driver.LEDS[i % NUM_LEDS] = pgm_read_dword(&(one[i]));
            //driver.LEDS<NEOPIXEL, 12>(leds, NUM_LEDS);
        }

    }
    long lastHandle = __clock_cycles();

    driver.showPixels();
    long lasthandle2 = __clock_cycles();
    Serial.printf("FPS fastled: %f \n", (float)240000000L / (lasthandle2 - lastHandle));
    start++;

    delay(2);
}`

hpwit, 2021-02-22:
try 
driver.LEDS[i % NUM_LEDS_PER_STRIP] = CRGB(pgm_read_dword(&(one[i])));
let me know

dallen98, 2021-02-22:
Hi Yves all I,m seeing is 8 leds lit up in red, I am using arduino ide 1.8.13, psram disabled, huge app (3mb No OTA 1mb Spiffs)

below is my code again with your ammendment on line 96 thanks

//#define FASTLED_INTERNAL //Elminates a compiler warning
#include <pgmspace.h> // Needed to store stuff in Flash using PROGMEM

//disable psram
//#define ESP32_VIRTUAL_DRIVER true //To enable virtual pin driver for esp32
//#define ESP_VIRTUAL_DRIVER_8 1 //to use the 8:1 ratio
//#define FULL_DMA_BUFFER
//#define NBIS2SERIALPINS 12 //12 pins max, any more is unstable
#define NBIS2SERIALPINS 1
//#define NBIS2SERIALPINS 10 //number of esp32 pins you will use. the total number of strips available will be NBIS2SERIALPINS * 8 here 72 strips
//#define NUM_LEDS_PER_STRIP 900 //the length of your strip if you have different strips size put here the longuest strip
#define LED_WIDTH 16
#define LED_HEIGHT 16
#define NUM_LEDS LED_WIDTH*LED_HEIGHT
#define NUM_LEDS_PER_STRIP 256 //16 x 16 ws2812b panel
#define MULTIPLE_LEDSBUFFER //use the mode several leds buffers in that mode the number of NBIS2SERIALPINS has to be <=12
#define STATIC_COLOR_GRB 1
#define USE_FASTLED //te enable the needed types
#include "FastLED.h" //this is the 8.1 yves version from https://github.com/hpwit/FASTLEDVIRTUALDRIVER
#include "I2SClocklessVirtualLedDriver.h"
#define LATCH_PIN 13
#define CLOCK_PIN 27 //for a reason I don't know, the CLOCK_PIN needs to be >=16

#define NUM_STRIPS NBIS2SERIALPINS * 8
#define NUM_LEDS NUM_LEDS_PER_STRIP *NUM_STRIPS

//int Pins[15] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4, 17, 16, 21, 23, 19};
//int Pins[12] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4, 17, 16}; //12 mins max
//int Pins[10] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4}; //12 pins max
int Pins[1] = {12}; //12 pins max
/*

    Define the esp pins you are using // 900 leds per pin max
    pin 12 will drive strip 1->8
    pin 14 will drive strip 9->16
    pin 26 will drive strip 17->24
    pin 25 will drive strip 25->32
    pin 33 will drive stip 33->40 //40 x 900=36000
    pin 32 will drive strip 41->48
    pin 5 will drive strip 49->56
    pin 15 will drive strip 57->64
    pin 2 will drive strip 65->72 // 72 x 900=64800
    pin 4 will drive strip 73->80 // 80 pins x 900 leds=72000 leds

                                // target 64800 (180 x 360= 64800)

    pin 17 will drive strip 81->88 //not required
    pin 16 will drive strip 89->96 //not required
    */

I2SClocklessVirtualLedDriver driver;
const unsigned long one[] PROGMEM = //this is 16x16 pixels "1" in blue on red background
{
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf80305, 0x83365c, 0x355998, 0xa52744, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xfe0000, 0xc6182a, 0x2062a8, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x046ebd, 0x46518b, 0xbd1d31, 0xf70306, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xa92640, 0x265fa3, 0x0070c0, 0x0070c0, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x2f5c9d, 0xa02a47, 0x365898, 0x7e3961, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xeb080e, 0x3a5694, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x3a5694, 0xeb080e, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xeb080e, 0x3a5694, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xf70406, 0x644474, 0x026fbe, 0x3a5694, 0xeb080e, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xeb080e, 0x3a5694, 0x026fbe, 0x644474, 0xf70406, 0xff0000, 0xff0000, 0xff0000, 0xff0000, 0xff0000,
[... 21 lines cut ...]
//driver.LEDS[i] = pgm_read_dword(&(one[i])); // Read array from Flash
//}
//int offset = i * NUM_LEDS_PER_STRIP; //this is the offest of the strip number i
for (int k = 0; k < (i+1); k++)
{
//because the strip is split between several led buffers this is how you can set a color
//driver.LEDS[k % NUM_LEDS_PER_STRIP + offset] = CHSV(k + start, 255, 255); //pin count rainbow
//driver.LEDS[i % NUM_LEDS_PER_STRIP] = pgm_read_dword(&(one[i]));
driver.LEDS[i % NUM_LEDS_PER_STRIP] = CRGB(pgm_read_dword(&(one[i])));   //yves

        //driver.LEDS[i % NUM_LEDS] = pgm_read_dword(&(one[i]));
        //driver.LEDS<NEOPIXEL, 12>(leds, NUM_LEDS);
    }

}
long lastHandle = __clock_cycles();

driver.showPixels();
long lasthandle2 = __clock_cycles();
Serial.printf("FPS fastled: %f \n", (float)240000000L / (lasthandle2 - lastHandle));
start++;

delay(2);

}

hpwit, 2021-02-22:
of course I am stupid lol
the loop was only going to display I stuff 
try this 
for (int i = 0; i < NUM_STRIPS; i++) {
int offset = i * NUM_LEDS_PER_STRIP;
for (int k = 0; k < NUM_LEDS_PER_STRIP; k++)
{
    driver.LEDS[k+offset] = CRGB(pgm_read_dword(&(one[k]))); //yves
}
}

dallen98, 2021-02-22:
yes works

dallen98, 2021-02-22:
I am now going to try and apply this to 4 16x16 panels arranged..
#include <FastLED_NeoMatrix.h>
#include <FastLED.h>

#include <LEDMatrix.h>
//
// Used by LEDMatrix
#define MATRIX_TILE_WIDTH   16 // width of EACH NEOPIXEL MATRIX (not total display)
#define MATRIX_TILE_HEIGHT  16 // height of each matrix
#define MATRIX_TILE_H       2  // number of matrices arranged horizontally
#define MATRIX_TILE_V       2  // number of matrices arranged vertically

// Used by NeoMatrix
#define mw (MATRIX_TILE_WIDTH *  MATRIX_TILE_H)
#define mh (MATRIX_TILE_HEIGHT * MATRIX_TILE_V)
#define NUMMATRIX (mw*mh)

// Compat for some other demos
#define NUM_LEDS NUMMATRIX
#define MATRIX_HEIGHT mh
#define MATRIX_WIDTH mw

Is this the best method to use ??? thanks

hpwit, 2021-02-23:
if you want to use this, you need to revert to not using the MULTILINE OPTION which should be ok if you're using 16x16 panels

dallen98, 2021-02-23:
My goal is to create a slide show of say 3 images, each image is 360x180 pixels, this image could be distributed over let us say 10 tiles each measuring 36x180 pixels or 12 tiles i,m flexible, I will of course convert this image below to an array which is 616kb
[image]

dallen98, 2021-02-23:
ps the 16x16 panels i spoke of is just for testing

hpwit, 2021-02-23:
ah oki
Then i would need to write something similar to their library for the mapping
how many pins per panel do you want ? 6 ?
regards

hpwit, 2021-02-23:
btw here the code is duplicating the same image on all 16*16 panels

dallen98, 2021-02-23:
My target 64800 (but I can be flexible) is (180 x 360= 64800) each of the 80 pins below could drive 900 leds
80pins x 900 leds =72000 leds total

pin 12 will drive strip 1->8
pin 14 will drive strip 9->16
pin 26 will drive strip 17->24
pin 25 will drive strip 25->32
pin 33 will drive stip 33->40 
pin 32 will drive strip 41->48
pin 5 will drive strip 49->56
pin 15 will drive strip 57->64
pin 2 will drive strip 65->72  
pin 4 will drive strip 73->80 // 80 pins x 900 leds=72000 leds

I want the 180 length because that is mostly what ws28xxx is supplied in so no soldering.

My panel size will always be 180 wide but the panel height does not need to be 36 high, I could use 12 pins..
int Pins[12] = {12, 14, 26, 25, 33, 32, 5, 15, 2, 4, 17, 16}; //12 pins max
pin 17 will drive strip 81->88 
pin 16 will drive strip 89->96 //96 pins x 675 leds =64800 leds

The problem with the above numbers is that given I am using 180 leds per length then the pin needs to land at the point of the connectors, it doesnt have to be 64800 leds, the width will always be 360px wide but the height could be a bit more than 180, I am mindfull that you say any more than 12 esp32 pins becomes unstable and the maximum leds on each virtual pin is about 900, sorry I,m very confused lol
thanks

what about a total size 360 x 192px, i use 12 pins (96 virtual pins) so 48 pins on each side, each pin drives 4 180 strips (720 leds) so each panel is 4px high x 180px wide 96 panels total

closed 2021-12-20 by hpwit