hpwit/FASTLEDVIRTUALDRIVER #7: Virtual Driver 8:1

midikidi, 2021-11-17:
The 5:1 driver 74hc595 pins are correct but 8:1 pin ordering appears to be wrong
as per your schematic looks like output 1 and 2 swapped with output 5 and 6

see test program 

#define ESP32_VIRTUAL_DRIVER true //To enable virtual pin driver for esp32
#define ESP_VIRTUAL_DRIVER_8 1 //to use the 8:1 ratio
#define NBIS2SERIALPINS 8 //number of esp32 pins you will use. the total number of strips available will be NBIS2SERIALPINS * 8 here 56 strips
#define NUM_LEDS_PER_STRIP 340 //the length of your strip if you have different strips size put here the longuest strip

#include "FastLED.h"
#define LATCH_PIN 4
#define CLOCK_PIN 21 //for a reason I don't know, the CLOCK_PIN needs to be >=16

#define NUM_STRIPS NBIS2SERIALPINS * 8
#define NUM_LEDS NUM_LEDS_PER_STRIP * NUM_STRIPS
CRGB leds[NUM_LEDS];

int Pins[NBIS2SERIALPINS] = { 13,14,25,26,33,32,15,27 };

void setup() {
	Serial.begin(115200);
	FastLED.addLeds<VIRTUAL_DRIVER, Pins, CLOCK_PIN, LATCH_PIN>(leds, NUM_LEDS_PER_STRIP);
	//FastLED.addLeds<VIRTUAL_DRIVER,Pins,CLOCK_PIN, LATCH_PIN,GRB>(leds,NUM_LEDS_PER_STRIP); //to set color order by default GRB
	FastLED.setBrightness(64);

}

void loop() {
	fill_solid(leds, 340, CRGB(255, 0, 0));//r
	fill_solid(leds + 340, 680, CRGB(0, 255, 0));//g
	fill_solid(leds + 680, 1020, CRGB(0, 0, 255));//b
	fill_solid(leds + 1020, 1360, CRGB(0, 255, 255));//cy
	fill_solid(leds + 1360, 1700, CRGB(255, 0, 255));//mag
	fill_solid(leds + 1700, 2040, CRGB(255, 255, 0));//yel
	fill_solid(leds + 2040, 2380, CRGB(255, 255, 255));//wh
	fill_solid(leds + 2380, 2410, CRGB(255, 0, 0));//r
	fill_solid(leds + 2720, 2760, CRGB(0, 255, 0));//g
	fill_solid(leds + 3060, 3090, CRGB(0, 0, 255));//g

	long     lastHandle = __clock_cycles();
	FastLED.show();
	long   lasthandle2 = __clock_cycles();
	Serial.printf("FPS fastled: %f \n", (float)240000000L / (lasthandle2 - lastHandle));

	delay(30);

}

hpwit, 2021-11-17:
Hello
I do not maintain this library anymore
You can use this one 
https://github.com/hpwit/I2SClocklessVirtualLedDriver
and for example
https://github.com/hpwit/I2SClocklessVirtualLedDriver/blob/main/examples/gettingstartedFastLED/gettingstartedFastLED.ino

Let me know if I can be of more help

midikidi, 2021-11-18:
Really struggling with the new driver.
bit of history i have designed 2 pcb's one for 51 one for 8:1
5:2 works fine as i mentioned before 8:1 mux ordering is wrong(copied your schematic)
As per your instructions I downloaded your new lib edited the pin numbers for my design.
if I do not include Fastled then GRGB is not declared
if I include Fastled I get compile errors

SAMPLE OF COMPILE ERRORS

fastspi_bitbang.h: 118:67: error: 'SPI_SPEED' was not declared in this scope
   #define CLOCK_HI_DELAY delaycycles<MIN_DELAY>(); delaycycles<(((SPI_SPEED-6) \ 2) - MIN_DELAY)>()
fastspi_bitbang.h:157: note  in expansion of macro CLOCK_HI_DELAY
   hi(clockpin); CLOCK_HI_DELAY
fastspi_bitbang.h: In static member function static void AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::writeBit(uint8_t, AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::clock_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::data_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::data_t, AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::data_t, AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::clock_t, AVRSoftwareSPIOutput<DATA_PIN, <anonymous> >::clock_t)

COPY OF TEST PRORAM

#define NBIS2SERIALPINS 4 //the number of virtual pins here mavimum 32 strips
#define STATIC_COLOR_GRB 1 //set the strip color
#define NUM_LEDS_PER_STRIP 340
#define NUM_LEDS (NUM_LEDS_PER_STRIP*NBIS2SERIALPINS*8)
#define CLOCK_PIN 21
#define LATCH_PIN 4
#define NUM_STRIPS 32

#include "FastLED.h"
#include "I2SClocklessVirtualLedDriver.h"

//here we have 3 colors per pixel
uint8_t leds[NUM_STRIPS*NUM_LEDS_PER_STRIP * 3];

int pins[16] = { 13,14,25,26 };

I2SClocklessVirtualLedDriver driver;
void setup() {
	Serial.begin(115200);
	driver.initled((uint8_t*)leds, pins, CLOCK_PIN, LATCH_PIN);
	driver.setBrightness(200);
}

void loop() {
	fill_solid(leds, 340, CRGB(255, 0, 0));//r
	fill_solid(leds + 340, 680, CRGB(0, 255, 0));//g
	fill_solid(leds + 680, 1020, CRGB(0, 0, 255));//b
	fill_solid(leds + 1020, 1360, CRGB(0, 255, 255));//cy
	fill_solid(leds + 1360, 1700, CRGB(255, 0, 255));//mag
	fill_solid(leds + 1700, 2040, CRGB(255, 255, 0));//yel
	fill_solid(leds + 2040, 2380, CRGB(255, 255, 255));//wh
	fill_solid(leds + 2380, 2410, CRGB(255, 0, 0));//r
	fill_solid(leds + 2720, 2760, CRGB(0, 255, 0));//g
	fill_solid(leds + 3060, 3090, CRGB(0, 0, 255));//g
	driver.showPixels();
	delay(30);
}

Please advise

hpwit, 2021-11-18:
here it is
`#include "FastLED.h"
#define NBIS2SERIALPINS 4 //the number of virtual pins here mavimum 32 strips
#define STATIC_COLOR_GRB 1 //set the strip color
#define NUM_LEDS_PER_STRIP 340
#define NUM_LEDS (NUM_LEDS_PER_STRIP * NBIS2SERIALPINS * 8)
#define CLOCK_PIN 21
#define LATCH_PIN 4
#define NUM_STRIPS 32

#include "FastLED.h"
#include "I2SClocklessVirtualLedDriver.h"

//here we have 3 colors per pixel
CRGB leds[NUM_LEDS];

int pins[4] = { 13,14,25,26 };

I2SClocklessVirtualLedDriver driver;
void setup() {
Serial.begin(115200);
driver.initled((uint8_t*)leds, pins, CLOCK_PIN, LATCH_PIN);
driver.setBrightness(200);
}

void loop() {
fill_solid(leds, 340, CRGB(255, 0, 0));//r
fill_solid(leds + 340, 680, CRGB(0, 255, 0));//g
fill_solid(leds + 680, 1020, CRGB(0, 0, 255));//b
fill_solid(leds + 1020, 1360, CRGB(0, 255, 255));//cy
fill_solid(leds + 1360, 1700, CRGB(255, 0, 255));//mag
fill_solid(leds + 1700, 2040, CRGB(255, 255, 0));//yel
fill_solid(leds + 2040, 2380, CRGB(255, 255, 255));//wh
fill_solid(leds + 2380, 2410, CRGB(255, 0, 0));//r
fill_solid(leds + 2720, 2760, CRGB(0, 255, 0));//g
fill_solid(leds + 3060, 3090, CRGB(0, 0, 255));//g
driver.showPixels();
delay(30);
}`

apparently the conflict came from CLOCK_PIN which is also declared in "fastled.h" if you put it before it seems to work fine in compilation.
I moved the include "fastled.h" above  or you can change CLOCK_PIN to CLOCKPIN for instance
let me know

midikidi, 2021-11-18:
Forget last post all sorted download latest version of fastled without all the 5:1 and 8:1 drivers added I2SClocklessVirtualLedDriver.h built demo sketch all pins of mux chip in the correct order. 
Can i still build code for  my 5:1 pcb design

hpwit, 2021-11-19:
Hello
Happy that it works
With this library you will not be able to do 5:1 I will need to look at it
But normally you should be able to compile your code while keeping the 5:1 8:1 fastled library that is my config.
may I asked what your are doing exactly ?

midikidi, 2021-11-20:
Hi Yves its David from Turkey www.ledsynth.com. We have had many conversation in the past under laserdave001 but all my accounts got screwed up. PS the new driver looks very clean and so do the ledstrips. Excellent Work

midikidi, 2021-11-20:
Yes I kept the old lib in the local project folder and one my 5:1 pcb's are used up its all 8:1

hpwit, 2021-11-20:
Hello David how are you ? glad to hear from you.
I hope you're doing well.
How can I help you more than that ?

midikidi, 2021-11-20:
Everything is fine thankyou . LedSynth is now running well 2 versions WS chipsets and Apa version. I have ditched rtos and using  different multitasking I can run 127 different tasks all writing to different parts of 17 ledstrips (and different animations).
I  have expanded the effects engine to include blend effects and split animations. the advantage of a split animation it only consumes one  task. I have written a pc app to configure the SD CARD. I have added patches which are a collection of effects so multiple strips/effects can be triggered at same time.

See below SD card config
{
  "ledstrip": 
    {
      "maxstrips": "20",
      "order": "RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB,RGB",
      "ledsperstrip": "90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90,90"
    }
  ,
  "ethernetadaptor": 
    {
      "portactive": "1",
      "ip": "192,168,2,100",
      "gw": "192,168,2,102",
      "sub": "192,168,2,101",
      "dns": "dns",
      "protocol": "artnet",
      "universe": "1",
      "channel": "2",
      "inport": "8100",
      "outport": "8200"
    }
  ,
  "wifiadaptor": 
    {
      "portactive": "1",
      "ssid": "ledsynth",
      "pw": "ledsynthnet",
      "ip": "192,168,1,101",
      "gw": "192,168,2,108",
      "sub": "192,168,2,109",
      "dns": "dns",
      "protocol": "osc",
      "universe": "1",
      "channel": "2",
      "inport": "8100",
      "outport": "7100"
    }
  ,
  "midi": 
    {
      "portactive": "1",
      "blackout": "1,81",
      "brightness": "1,48",
      "patch": "1,0",
      "scene": "2,0",
      "strip": "1,49",
      "effect": "1,50",
      "direction": "1,51",
      "start": "1,52",
      "length": "1,53",
      "speed": "1,54"
    }
  ,
  "portal": 
    {
      "portactive": "1",
      "ssid": "LedsynthPro",
      "pw": ""
    }
  ,
  "dmx": 
    {
      "portactive": "1",
      "startaddress": "2"
    }
  ,
[... 233 lines cut ...]
      "patch42": "21,22,23",
      "patch43": "13",
      "patch44": "14,15,16,17,18",
      "patch45": "19,20",
      "patch46": "21,22,23",
      "patch47": "1,2,3,4,5,6,7,8",
      "patch48": "9,10,11,12",
      "patch49": "14,15,16,17,18",
      "patch50": "19,20",
      "patch51": "21,22,23",
      "patch52": "1,2,3,4,5,6,7,8",
      "patch53": "9,10,11,12",
      "patch54": "13",
      "patch55": "14,15,16,17,18",
      "patch56": "19,20",
      "patch57": "21,22,23",
      "patch58": "13",
      "patch59": "14,15,16,17,18",
      "patch60": "19,20",
      "patch61": "21,22,23",
      "patch62": "1,2,3,4,5,6,7,8",
      "patch63": "9,10,11,12"
    }
  
}

hpwit, 2021-11-20:
Nice work !!!
I hope to be able to see it running.
Are you implementing the 5:1 or 8;1 library in your product ?

midikidi, 2021-11-20:
For the first few units i will use 5:1(to use up old stock) but i designed a new driver 8:1 with rs485 differential outputs
also apa102 rs485 i have rs485 receiver board for apa and ws i am test with over 60 metres cat 5  cable

hpwit, 2021-11-20:
Ah cool I am happy to see that what I've done is used in a more professional environment.
rs485 is nice to go far away.

midikidi, 2021-11-21:
Can you please point me to a url for latest APA102 drivers

midikidi, 2021-11-21:
See below APA102 CRASH APP AND DEBUG

/*
 Name:		APA102Test.ino
 Created:	10/15/2021 8:14:34 PM
 Author:	LaserLaptop
 //*********************************************************************************************************************
 //CRASH DETAILS
 Backtrace: 0x4008caaf:0x3ffb1e10 0x40086e82:0x3ffb1e20 0x400d11a8:0x3ffb1ee0 0x400d13cc:0x3ffb1f60 0x400d13d7:0x3ffb1f90 0x400d2089:0x3ffb1fb0 0x4008a7c1:0x3ffb1fd0

Rebooting...
f[06]~[1E][06][18]†f~àž[00][18]x†[18][18]x[00]˜à˜ø[18]˜à[18]x˜[18]x[06][18]~`[18][1E]æ€˜€æ€˜€[18]æø[06]~˜[18][18]æà[06][18]†[1E]ffþf€[06]ffæf˜f`[1E]f[06][06]f††à†[18]æžþž`†~[00][18]f˜x[00][18]†[1E]ff[18]æf`[06]æ˜[06]fþf`à˜[06]æ˜[06]˜þf€æ˜þ˜`f˜˜[1E]˜€[1E]æžøžxfž~æøfž[00]†~[00][18][18]†[00]˜˜[06]f[18]æf[00]f~[00][18]fxff€˜€[1E][06]žžžþfx[18]†~˜[18][18][06][18][18]†[06]æfx[18]†~˜[18][18][06][18][18]†`æfx[18]†~˜[18][18][06][18][18]†[1E]æø[06]˜fx[18]†~˜[18][18][06][18][18]†€ž`æfx[18]†~˜[18][18][06][18][18]†~[06]æfx[18]†~˜[18][18][06][18]˜[1E]˜€æžþž`fx˜[18]`†˜þ˜à†[00]˜à[06]žþž[1E]ææ[06][18]x†žx†~[06]˜[1E]˜€àžþž[06][06]x˜[18][18]æxx†xx[06][18]˜à[18][18]†àžf†ž˜[18][1E]æ€˜€àžþž[06][06]x˜[18][18]æxx†xx[06][18]˜à˜à[06]†àžf†ž˜˜à[18]x[06][18][1E]æ€˜€€žþž[00][18][18][18]~[06]fžàž[00]˜à[18]x[00][18]øæžþžæž[00][18][1E]æ€˜€àžþž[06][06]x˜[18][18][06][1E][00][18]˜[1E]€[18][18][18][18]†àžf†ž˜˜ø˜[1E][18][18]˜[1E]˜€€žþž[00][18][18][18]~[06]fžàž[00]˜à[18]x[00][18]øæžþžæž[00][18][1E]æ€˜€àžþž[06][06]x˜[18][18][06][1E][00][18][18][18][18][1E][00][18][18]†àžf†ž˜[18][1E][1E]˜æ[18]xæ€˜€f†ž`†øf[06][18][18][06][1E][00][18][18][18][18][1E][18][06]˜[1E]˜€in d
in d1
in d2
in d3
in d4
in d5
100000
in d6
in d7
61
42
-4
31
div_b 31
bck_div_num 8
in d5
in d6
//*****************************************************************************************************************************
*/

#include "Log.h"
#include "FastLED.h"

#define NUM_LEDS_PER_STRIP 340
#define NUM_STRIPS 9 
#define NUM_LEDS NUM_STRIPS * NUM_LEDS_PER_STRIP
#define CLOCK_PIN 21 //needs to be higher than 16
#include "I2SAPA102.h"

#define FRAMES_PER_SECOND  120
int Pins[9] = {4,13,14,25,26,33,32,15,27 };

CRGB leds[NUM_LEDS];
I2SAPA102 controller(0);

void setup() {
	Serial.begin(230400);
	controller.initled(leds,Pins,CLOCK_PIN,NUM_STRIPS,NUM_LEDS_PER_STRIP,2); //for 2MHZ
	fill_solid(leds, 100, CRGB::Black);
	controller.showPixels();
	controller.setBrightness(31); //to be used instead of fastled.setbritghness value 0->31 the default ligtning of the apa102
	delay(200);
	controller.showPixels();

}

void loop() {
	fill_solid(leds, 339, CRGB(255, 0, 0));//r
	fill_solid(leds + 340, 679, CRGB(0, 255, 0));//g
	fill_solid(leds + 680, 1019, CRGB(0, 255, 0));//b
	fill_solid(leds + 1020, 1359, CRGB(0, 255, 255));//cy
	fill_solid(leds + 1360, 1699, CRGB(255, 0, 255));//mag

	//**************************************************************
	//causes irratic behavour core error crash leds size 9*340(3060)
	fill_solid(leds + 1700, 2039, CRGB(255, 255, 0));//yel
	fill_solid(leds + 2040, 2379, CRGB(255, 255, 255));//wh

	controller.showPixels();
	delay(50);
}

hpwit, 2021-11-21:
you've recieved an invitation use the script in HD107 example

midikidi, 2021-11-21:
hd107 script ??

midikidi, 2021-11-21:
Sorry to bombard you just wrapping up code/product
the new 8:1 driver causes  the esp32 to restart every minute or so if i disable wifi no reboot.
ledsynth uses wifi and or ethernet, using latest esp32 psram silicon any ideas

hpwit, 2021-11-21:
Did you get the error on the serial output ? And did you decode it using eps error decoder ?

hpwit, 2021-11-21:
hd107 script ??
you have accès to a new repository
https://github.com/hpwit/I2SHD108
In the example use the one linked to hd107

midikidi, 2021-11-21:
this is a dump of crash

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40159adc  PS      : 0x00060034  A0      : 0x800818b0  A1      : 0x3ffbe710  
A2      : 0x00000015  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x0000270f  
A6      : 0x00000154  A7      : 0x3ffb2068  A8      : 0x8008132c  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x0000270f  A12     : 0x3ffba264  A13     : 0x0000abab  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008b61f  EPC2    : 0x00000000  EPC3    : 0x40089d6f  EPC4    : 0x40159adc

Backtrace: 0x40159adc:0x3ffbe710 0x400818ad:0x3ffbe790 0x40087c01:0x3ffbe7d0 0x40089d6c:0x3ffba190 0x40085ecb:0x3ffba1b0 0x4008bbdd:0x3ffba1d0

Rebooting...
f[06]~[1E][06][18]†f~àž[00][18]x†[18][18]x[00]˜à˜ø[18]˜à[18]x˜[18]x[06][18]~`[18][1E]æ€˜€æ€˜€[18]æø[06]~˜[18][18]æx[00][18]†[1E]ffþf˜f`[1E]f[06][06]f††à†[18]æžþž`†~[00][18]f˜x[00][18]†[1E]ff[18]æf`[06]æ˜[06]fþf`à˜[06]æ˜[06]˜þf€æ˜þ˜`f˜˜[1E]˜€[1E]æžøžxfž~æøfž[00]†~[00][18][18]†[00]˜˜[06]f[18]æf[00]f~[00][18]fxff€˜€[1E][06]žžžþfx[18]†~˜[18][18][06][18][18]†[06]æfx[18]†~˜[18][18][06][18][18]†`æfx[18]†~˜[18][18][06][18][18]†[1E]æø[06]˜fx[18]†~˜[18][18][06][18][18]†€ž`æfx[18]†~˜[18][18][06][18][18]†~[06]æfx[18]†~˜[18][18][06][18]˜[1E]˜€æžþž`fx˜[18]`†˜þ˜à†[00]˜à[06]žþž[1E]ææ[06][18]x†žx†~[06]˜[1E]˜€àžþž[06][06]x˜[18][18]æxx†xx[06][18]˜à[18][18]†àžf†ž˜[18][1E]æ€˜€àžþž[06][06]x˜[18][18]æxx†xx[06][18]˜à˜à[06]†àžf†ž˜˜à[18]x[06][18][1E]æ€˜€€žþž[00][18][18][18]~[06]fžàž[00]˜à[18]x[00][18]øæžþžæž[00][18][1E]æ€˜€àžþž[06][06]x˜[18][18][06][1E][00][18]˜[1E]€[18][18][18][18]†àžf†ž˜˜ø˜[1E][18][18]˜[1E]˜€€žþž[00][18][18][18]~[06]fžàž[00]˜à[18]x[00][18]øæžþžæž[00][18][1E]æ€˜€àžþž[06][06]x˜[18][18][06][1E][00][18][18][18][18][1E][00][18][18]†àžf†ž˜[18][1E][1E]˜æ[18]xæ€˜€f†ž`†øf[06][18][18][06][1E][00][18][18][18][18][1E][18][06]˜[1E]˜€ÿWaiting for Wifi to connect
***************
Network Not Connected
Check SD Card for Correct SSID and Password
All WiFi Protocols are Disabled

LEDSynth Active

hpwit, 2021-11-22:
Hello
You need to put 
Backtrace: 0x40159adc:0x3ffbe710 0x400818ad:0x3ffbe790 0x40087c01:0x3ffbe7d0 0x40089d6c:0x3ffba190 0x40085ecb:0x3ffba1b0 0x4008bbdd:0x3ffba1d0
in the esp error decoder 
This option should be available in ‘tools’=> esp exception decoder. 
I can’t do it because it needs to be run where the program has been compiled and run

midikidi, 2021-11-22:
New to exception decoder installed for arduino ide (which i dont use) But this is what Visual micro gave me
Exception:Error:
C:\SOFTWA~1\LEDSYN~1\LEDSYN~2.00/I2SClocklessVirtualLedDriver.h(175): error 0x40159ee4:MOD(int, int)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996:::0x4008b61f:xQueueGenericSend
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089d61:spi_flash_op_block_func
C:\Users\LaserLaptop\AppData\Local\Temp\VMBuilds\LEDsynthPro81-V2\esp32_esp32wrover\Debug\effectsengine/Effectlookup.h(176): error 0x400818ad:_I2SClocklessVirtualLedDriverinterruptHandler(void*)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323:::0x40087c01:_xt_medint3
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:81:::0x40089d5e:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62:::0x40085ecb:ipc_task
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x4008bbdd:vPortTaskWrapper

not sure if it helps

hpwit, 2021-11-23:
Hello
I see that there is a pointer to your code. I would need the full code to do full debugging
Effectlookup.h for instance

midikidi, 2021-11-23:
#pragma once
//effect enums

uint8_t geteffectindex(String fn);

const char * const effectlist[] PROGMEM = {
"Rainbow",
"RainbowWithGlitter",
"Confetti",
"Sinelon",
"Juggle",
"Bpm",
"Pride",
"Threemovingcolors",
"Fire",
"Dobeatwave",
"DoBlur",
"HeartBeat",
"HeartBeat1",
"HeartBeat2",
"Fadetowardssolid",
"Fadeupdown",
"Pacifica",
"Fullstrobeloop",
"Split_anim1",
"Split_anim2",
"Split_anim3",
"Split_anim4",
"Split_anim5",
"Split_anim6",
"Split_anim7",
"Split_anim8",
"Split_anim9",
"Split_anim10",
"Split_anim11",
"Split_anim12",
"Split_anim13",
"Split_anim14",
"Split_anim15",
"Split_anim16",
"Split_anim17",
"Split_anim18",
"Split_anim19",
"Split_anim20",
"Split_anim21",
"Split_anim22",
"Split_anim23",
"Split_anim24",
"Split_anim25",
"Split_anim26",
"Split_anim27",
"Split_anim28",
"Split_anim29",
"Split_anim30",
"Split_anim31",
"Split_anim32",
"Blend_1",
"Blend_2",
"Blend_3",
"Blend_4",
"Blend_5",
"Blend_6",
"Blend_7",
"Blend_8",
"Blend_9",
"Blend_10",
"Fillred",
"Fillgreen",
"Fillblue",
"Fillcyan",
"Fillmagenta",
"Fillyellow",
"Fillwhite",
"Fillblack",
"Palette1",
"Palette2",
"Palette3",
"Palette4",
"Palette5",
"Palette6",
"Palette7",
"Palette8",
"Palette9",
"Palette10",
"Palette11",
"Palette12",
"Palette13",
"Palette14",
"Palette15",
"Palette16",
"Palette17",
"Palette18",
"Palette19",
"Palette20",
"Palette21",
"Palette22",
"Palette23",
"Palette24",
"Palette25",
"Palette26",
"Palette27",
"Palette28",
"Palette29",
"Palette30",
"Palette31",
"Palette32",
"Palette33",
"Palette34",
"Palette35",
"Palette36",
"Palette37",
"Palette38",
"Palette39",
"Palette40",
"Palette41",
"Palette42",
"Palette43",
"Palette44",
"Palette45",
"Palette46",
"Palette47",
"Palette48",
"Palette49",
"Palette50",
"Palette51",
"Palette52",
"Palette53",
"Palette54",
"Palette55",
"Palette56",
"Palette57",
"Palette58",
"Palette59",
"Palette60",
"Palette61",
"Palette62",
"Palette63",
"Palette64",
"Palette65",
"Palette66",
"Palette67",
"Palette68",
"Palette69",
"Palette70",
"Palette71",
"Palette72",
"Palette73",
"Palette74",
"Palette75",
"Palette76",
"Palette77",
"Palette78",
"Palette79",
"Palette80",
"Palette81",
"Palette82",
"Palette83",
"Palette84",
"Palette85",
"Palette86",
};

uint8_t geteffectindex(String name)
{

	for (int i = 0; i < MAXEFFECTS; i++)
		if (strcmp(name.c_str(), effectlist[i]) == 0)
			return i;
}

uint8_t geteffectlistsize()
{
	int size = sizeof(effectlist) / sizeof(int);
	return size;
}

midikidi, 2021-11-23:
interesting same codebase for 5:1 mux driver
line 176 is the return of the geteffectlistsize

hpwit, 2021-11-23:
you have an untreated case in your geteefectlist  if you do not find what you 're looking for then ... I don't know what happens it should be something like this
``
uint8_t geteffectindex(String name)
{

for (int i = 0; i < MAXEFFECTS; i++)
{
  if (strcmp(name.c_str(), effectlist[i]) == 0)
    return i;
}
 
 return MAXEFFECTS;
}
``

and in your calling code you should have
effectnum=geteffectindex("xxx");
if(effectnum== MAXEFFECTS)
   print("error");
esle
  playeffect

midikidi, 2021-11-23:
well spotted this shouldn't happen unless the config file is wrong which is created by my c# app doest crash with wifi disabled very interesting.

hpwit, 2021-11-23:
I think this is a conflict between all the tasks.
I would need to see the full code to  see what happens exactly.
does it break always at the same point ?

midikidi, 2021-11-23:
I will do some more testing, im no longer using rtos, it could be the DMX library . happy to send code but project is massif

midikidi, 2021-11-23:
I will investigate what tasks im using allocating

midikidi, 2021-11-24:
8:1 MUX Driver integrated into LEDSynth WIFI active been running 1 hour no reboot. It was indeed the bug in effects lookup
causing the issue as I said it shouldn't happen as config file automatically created by my  C# app. However this was an old config file manually created and I misspelt Fillwhite as Fillwite. Lookup error fixed and storing effec now checks for valid.

void addeffect(uint8_t strip, uint8_t fxno, uint16_t sp, uint16_t len, bool dir, uint8_t spd, uint8_t mode)
{
	if (fxno != MAXEFFECTS) {
		effect[allocatedtasks].effect_number = fxno;
		effect[allocatedtasks].strip_number = strip;
		effect[allocatedtasks].effect_startposition = sp;
		effect[allocatedtasks].effect_length = len;
		effect[allocatedtasks].effect_direction = dir;
		effect[allocatedtasks].effect_speed = spd;
		effect[allocatedtasks].effect_mode = mode;
		allocatedtasks++;
	}
Many thanks for your help
just HD107 to sort

hpwit, 2021-11-24:
Good to hear
indeed when a function is not correctly finished then who knows where these are going
if you can show me how it's looks like ;:)

midikidi, 2021-11-25:
Spoke to soon 8:1 mux still crashing about every 3 minutes. I produced a hacked down version of LEDSynth stripped out most of the code and  librarys. Attached is the code and exception decode

Exception:Error:
C:\SOFTWA~1\LEDSYN~1\LEDSYN~1.00-/I2SClocklessVirtualLedDriver.h(175): error 0x401435d8:MOD(int, int)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996:::0x4008b6d1:xQueueGenericReceive
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089a9d:spi_flash_op_block_func
C:\Software Development\LEDSYNTH-28102021\LEDSynthPro-81-V12.00 - DEBUGHACK\multitasking/MultiTasking.h(1513): error 0x400818ad:_I2SClocklessVirtualLedDriverinterruptHandler(void*)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323:::0x4008793d:_xt_medint3
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:81:::0x40089a9a:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62:::0x40085c07:ipc_task
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x4008b919:vPortTaskWrapper

for some reason this site wont allow me to upload the zipped project it says it supports ZIP files

closed 2021-11-25 by midikidi