hpwit/FASTLEDVIRTUALDRIVER #8: APA 102 DRIVER

midikidi, 2021-11-21:
Rebooting...
f[06][1E][06][18]†fàž[00][18]x†[18][18]x[00]˜à˜ø[18]˜à[18]x˜[18]x[06][18][18][1E]æ€˜€æ€˜€[18]æø[06]~˜[18][18]æà[06][18]†[1E]ffþf€[06]ffæf˜f[1E]f[06][06]f††à†[18]æžþž†~[00][18]f˜x[00][18]†[1E]ff[18]æf[06]æ˜[06]fþfà˜[06]æ˜[06]˜þf€æ˜þ˜f˜˜[1E]˜€[1E]æžøžxfžæøfž[00]†[00][18][18]†[00]˜˜[06]f[18]æf[00]f[00][18]fxff€˜€[1E][06]žžžþfx[18]†˜[18][18][06][18][18]†[06]æfx[18]†˜[18][18][06][18][18]†æfx[18]†~˜[18][18][06][18][18]†[1E]æø[06]˜fx[18]†~˜[18][18][06][18][18]†€žæfx[18]†˜[18][18][06][18][18]†[06]æfx[18]†˜[18][18][06][18]˜[1E]˜€æžþžfx˜[18]†˜þ˜à†[00]˜à[06]žþž[1E]ææ[06][18]x†žx†[06]˜[1E]˜€àžþž[06][06]x˜[18][18]æxx†xx[06][18]˜à[18][18]†àžf†ž˜[18][1E]æ€˜€àžþž[06][06]x˜[18][18]æxx†xx[06][18]˜à˜à[06]†àžf†ž˜˜à[18]x[06][18][1E]æ€˜€€žþž[00][18][18][18][06]fžàž[00]˜à[18]x[00][18]øæžþžæž[00][18][1E]æ€˜€àžþž[06][06]x˜[18][18][06][1E][00][18]˜[1E]€[18][18][18][18]†àžf†ž˜˜ø˜[1E][18][18]˜[1E]˜€€žþž[00][18][18][18][06]fžàž[00]˜à[18]x[00][18]øæžþžæž[00][18][1E]æ€˜€àžþž[06][06]x˜[18][18][06][1E][00][18][18][18][18][1E][00][18][18]†àžf†ž˜[18][1E][1E]˜æ[18]xæ€˜€f†ž`†øf[06][18][18][06][1E][00][18][18][18][18][1E][18][06]˜[1E]˜€in d
in d1
in d2
in d3
in d4
in d5
100000
in d6
in d7
61
42
-4
31
div_b 31
bck_div_num 8
in d5
in d6
//*****************************************************************************************************************************
*/

#include "Log.h"
#include "FastLED.h"

#define NUM_LEDS_PER_STRIP 340
#define NUM_STRIPS 9
#define NUM_LEDS NUM_STRIPS * NUM_LEDS_PER_STRIP
#define CLOCK_PIN 21 //needs to be higher than 16
#include "I2SAPA102.h"

#define FRAMES_PER_SECOND 120
int Pins[9] = {4,13,14,25,26,33,32,15,27 };

CRGB leds[NUM_LEDS];
I2SAPA102 controller(0);

void setup() {
Serial.begin(230400);
controller.initled(leds,Pins,CLOCK_PIN,NUM_STRIPS,NUM_LEDS_PER_STRIP,2); //for 2MHZ
fill_solid(leds, 100, CRGB::Black);
controller.showPixels();
controller.setBrightness(31); //to be used instead of fastled.setbritghness value 0->31 the default ligtning of the apa102
delay(200);
controller.showPixels();

}

void loop() {
fill_solid(leds, 339, CRGB(255, 0, 0));//r
fill_solid(leds + 340, 679, CRGB(0, 255, 0));//g
fill_solid(leds + 680, 1019, CRGB(0, 255, 0));//b
fill_solid(leds + 1020, 1359, CRGB(0, 255, 255));//cy
fill_solid(leds + 1360, 1699, CRGB(255, 0, 255));//mag

//**************************************************************
//causes irratic behavour core error crash leds size 9*340(3060)
fill_solid(leds + 1700, 2039, CRGB(255, 255, 0));//yel
fill_solid(leds + 2040, 2379, CRGB(255, 255, 255));//wh

controller.showPixels();
delay(50);
}

midikidi, 2021-11-21:
Can you please point me to a url for latest APA102 drivers

midikidi, 2021-11-21:
thanks got the invitation is 107 and apa102 timing same

hpwit, 2021-11-21:
It’s not a matter of timing but matter of data structure :)
Yes they are the same

midikidi, 2021-11-23:
New basic code for hd107 build and tested now to test with wifi then integration to ledsynth main engine. I love the ability to set an individual pixel brightness Im sure this opens up posibilities for new type effects

hpwit, 2021-11-23:
this is not the final version I have another one a bit better but I need to polish it

midikidi, 2021-11-23:
Please keep me informed. I am also helping a friend with a product he manufactures IP rated pixel strips aluminium extrusion with internal power supply original APA102 he adds optics to the pixels to tighten the beam. I have a unit here for test at 4mhz clock there are issues. If I set the speed to 2mhz all is ok. I need to investigate further as it could be my RS485 Hardware. In your driver did you do the timing from scratch 
[image]

midikidi, 2021-11-24:
Prior to integration I setup this simple sketch, unless I am doing something stupid there is no output on 
pins26,33,32,15,27,12. pins 4,13,14,25 no issue

/*
 Name:		HD107NEWDRIVER.ino
 Created:	11/21/2021 11:23:36 PM
 Author:	LaserLaptop
*/

#include "FastLED.h"

#define NUM_LEDS_PER_STRIP 40
#define NUM_STRIPS 10
#define NUM_LEDS NUM_STRIPS * NUM_LEDS_PER_STRIP
#define CLOCK_PIN 21 //needs to be higher than 16
#include "I2SHD108.h"

int Pins[16] = { 4,13,14,25,26,33,32,15,27,12 };

HD107Led leds[NUM_LEDS];
CRGB workingbuffer[NUM_LEDS_PER_STRIP];
I2SHD108 controller(0);

//FORWARD DECS
void RGBfill_workingbuffer(int start, int end, int r, int g, int b, bool order, uint8_t speed);
void RGBfill_ledsbuffer(int stripno, int start, int end, int r, int g, int b, bool order, uint8_t speed);

void RGBfill_workingbuffer(int start, int end, int r, int g, int b, bool order, uint8_t speed)
{

	for (int j = start; j < end; j++)
	{
		workingbuffer[j] = CRGB(r, g, b);
	}
}

void RGBfill_ledsbuffer(int stripno, int start, int end, int r, int g, int b, bool order, uint8_t speed)
{
	int stripoffset = stripno * NUM_LEDS_PER_STRIP;
	for (int j = start; j < end; j++)
	{
		leds[j + stripoffset] = CRGB(r, g, b);
	}
}

void striptest() {

	RGBfill_ledsbuffer(0, 0, 10, 255, 0, 0, true, 0);
	RGBfill_ledsbuffer(0, 10, 20, 0, 255, 0, true, 0);
	RGBfill_ledsbuffer(0, 20, 30, 0, 0, 255, true, 0);

	RGBfill_ledsbuffer(1, 0, 10, 0, 255, 255, true, 0);
	RGBfill_ledsbuffer(1, 10, 20, 255, 0, 255, true, 0);
	RGBfill_ledsbuffer(1, 20, 30, 255, 255, 0, true, 0);

	RGBfill_ledsbuffer(2, 0, 10, 255, 0, 0, true, 0);
	RGBfill_ledsbuffer(2, 10, 20, 0, 255, 0, true, 0);
	RGBfill_ledsbuffer(2, 20, 30, 0, 0, 255, true, 0);

	RGBfill_ledsbuffer(3, 0, 10, 0, 255, 255, true, 0);
	RGBfill_ledsbuffer(3, 10, 20, 255, 0, 255, true, 0);
	RGBfill_ledsbuffer(3, 20, 30, 255, 255, 0, true, 0);

	RGBfill_ledsbuffer(4, 0, 10, 255, 0, 0, true, 0);
	RGBfill_ledsbuffer(4, 10, 20, 0, 255, 0, true, 0);
	RGBfill_ledsbuffer(4, 20, 30, 0, 0, 255, true, 0);

	RGBfill_ledsbuffer(5, 0, 10, 0, 255, 255, true, 0);
	RGBfill_ledsbuffer(5, 10, 20, 255, 0, 255, true, 0);
	RGBfill_ledsbuffer(5, 20, 30, 255, 255, 0, true, 0);

	RGBfill_ledsbuffer(6, 0, 10, 255, 0, 0, true, 0);
	RGBfill_ledsbuffer(6, 10, 20, 0, 255, 0, true, 0);
	RGBfill_ledsbuffer(6, 20, 30, 0, 0, 255, true, 0);

	RGBfill_ledsbuffer(7, 0, 10, 0, 255, 255, true, 0);
	RGBfill_ledsbuffer(7, 10, 20, 255, 0, 255, true, 0);
	RGBfill_ledsbuffer(7, 20, 30, 255, 255, 0, true, 0);

	RGBfill_ledsbuffer(8, 0, 10, 255, 0, 0, true, 0);
	RGBfill_ledsbuffer(8, 10, 20, 0, 255, 0, true, 0);
	RGBfill_ledsbuffer(8, 20, 30, 0, 0, 255, true, 0);

	RGBfill_ledsbuffer(9, 0, 10, 0, 255, 255, true, 0);
	RGBfill_ledsbuffer(9, 10, 20, 255, 0, 255, true, 0);
	RGBfill_ledsbuffer(9, 20, 30, 255, 255, 0, true, 0);

	//leds[0] = Brightness(2);

	long ts2 = __clock_cycles();
	controller.showPixels();
	Serial.printf("fps:%f\n", (float)240000000 / (__clock_cycles() - ts2));
	delay(400);
}

void setup() {
	Serial.begin(230400);
	controller.initled((uint8_t*)leds, Pins, CLOCK_PIN, NUM_STRIPS, NUM_LEDS_PER_STRIP, HD107, 2); //default Speed 2MHz 
	//controller.initled(leds,Pins,CLOCK_PIN,NUM_STRIPS,NUM_LEDS_PER_STRIP,freq in MHZ);
	//controller.initled(leds,Pins,CLOCK_PIN,NUM_STRIPS,NUM_LEDS_PER_STRIP,2); //for 2MHZ
	delay(200);

	//set the brightness to max for all of the leds
	for (int i = 0; i < NUM_STRIPS*(NUM_LEDS_PER_STRIP); i++)
	{
		//this affect only the brightness of the leds
		leds[i] = Brightness(31);
	}
}

void loop() {
	striptest();
}

hpwit, 2021-11-24:
on line 
35 of file of I2SHD108. h
 add
#define NUMSTRIPS 16
 line 39

hpwit, 2021-11-24:
> Please keep me informed. I am also helping a friend with a product he manufactures IP rated pixel strips aluminium extrusion with internal power supply original APA102 he adds optics to the pixels [...]

Hello on that part it’s not your rs485 it’s the speed of the esp32. In the new library I am working on being able to push at 4mhz maybe more. I need to work on it a bit more

midikidi, 2021-11-26:
wheres the latest hd10 fix cant see it in shared repository ?

midikidi, 2021-11-27:
SORRY, misunderstanding I thought you had release an update to the hd108 driver. yes the #define fixed that problem all 10 RS485 outputs working

hpwit, 2021-11-29:
Hello
Sorry I did not have the time to look at it. I will have a look later today or tomorrow

midikidi, 2021-12-01:
The problem of the LEDSynth (HD107 DRIVER) is Hardware and maybe your driver as YOU pointed out.
At 1mhz I can drive APA102 over screened CAT5 RS485 transmitter --> receiver . If I use an RJ45 female to female unscreened joiner and use a section of unscreened CAT5 I get errors. I will get a longer length of  Screened cat 5 and test again

midikidi, 2021-12-01:
Update using current version of your hd driver clocked at 1mhz driving apa102 at 37metres away
 no screened cat5 a poor quality joiner linking 2 cat5 cables

hpwit, 2021-12-01:
Hello is it working well ?

midikidi, 2021-12-01:
JUST AN OCCASIONAL GLITCH THEN IT CORRECTS ITS SELF WONT RUN FASTER THAN 1MHZ RELIABLY
im using 37mts unshielded cat 5 so could still he a noise issue even as it is twisted pair will continue to do tests will try and pickup a very long shielded cat 5

hpwit, 2021-12-02:
hello good to hear.
I am busy with a critical go to production at work I will get to it asap

midikidi, 2021-12-02:
Thanyou Yves

closed 2021-12-20 by hpwit