hpwit/FASTLEDVIRTUALDRIVER #10: MUX 8:1 DRIVER

midikidi, 2021-11-25:
Hi Yves 8:1 issue is not closed Github crashed when I tried to upload zip file
reboot happens about every 3 minutes

Exception:Error:
C:\SOFTWA~1\LEDSYN~1\LEDSYN~1.00-/I2SClocklessVirtualLedDriver.h(175): error 0x401435d8:MOD(int, int)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996:::0x4008b6d1:xQueueGenericReceive
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089a9d:spi_flash_op_block_func
C:\Software Development\LEDSYNTH-28102021\LEDSynthPro-81-V12.00 - DEBUGHACK\multitasking/MultiTasking.h(1513): error 0x400818ad:_I2SClocklessVirtualLedDriverinterruptHandler(void*)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323:::0x4008793d:_xt_medint3
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:81:::0x40089a9a:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62:::0x40085c07:ipc_task
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x4008b919:vPortTaskWrapper

Dave

hpwit, 2021-11-25:
hello in the I2SClocklessVirtualLedDriver.h
line 174
`int MOD(int a, int b)
{`
=>
`int MOD(int a, int b)
{
if(b==0)
return a;

`

midikidi, 2021-11-25:
driver changed
int MOD(int a, int b)
{

   if(b==0)
    {

         return a;
    }
    
    if (a < 0)
    {
        if (-a % b == 0)
            return 0;
        else
            return b - (-a) % b;
    }
    else
        return a % b;
}

Exception:Error:
C:\SOFTWA~1\LEDSYN~1\LEDSYN~1.00-/I2SClocklessVirtualLedDriver.h(175): error 0x401435ac:MOD(int, int)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089aa3:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089aa8:spi_flash_op_block_func
C:\Software Development\LEDSYNTH-28102021\LEDSynthPro-81-V12.00 - DEBUGHACK\multitasking/MultiTasking.h(1513): error 0x400818ad:_I2SClocklessVirtualLedDriverinterruptHandler(void*)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323:::0x4008793d:_xt_medint3
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089aa5:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62:::0x40085c07:ipc_task
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x4008b919:vPortTaskWrapper

midikidi, 2021-11-25:
code via dropbox
https://www.dropbox.com/sh/q1fjzaohbdbfids/AABU721QdeYKnwvz2ahrTcsua?dl=0

midikidi, 2021-11-26:
GETING CLOSER TO TRACKING THIS BUG DOWN, It would appear if wifi connects all is ok. If wifi fails to connect it reboots occassionaly. im curious what version of esp32 core are you using

hpwit, 2021-11-26:
hello
try to do this
static int IRAM_ATTR  MOD(int a, int b)
instead of 
int MOD(int a, int b)

because I think a conflict accessing the data
i am using Arduino to compile
BTW: have you seen to correction for the HD108.h

midikidi, 2021-11-26:
Will try fix if i setup driver but in loop dont do any thing no crash , didnt see hd correction

midikidi, 2021-11-26:
crash
Exception:Error:
C:\Users\LaserLaptop\AppData\Local\Temp\VMBuilds\LEDsynthPro81-V2\esp32_esp32wrover\Release\multitasking/TaskScheduler.h(543): error 0x400d1628:I2SClocklessVirtualLedDriver
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323:::0x40087948:_xt_medint3
C:\Users\LaserLaptop\AppData\Local\Temp\VMBuilds\LEDsynthPro81-V2\esp32_esp32wrover\Release\multitasking/TaskScheduler.h(543): error 0x40081724:_ZL16loadAndTransposePh13OffsetDisplayPtiS_S_S_S_S_S_S_$isra$1
C:\Users\LaserLaptop\AppData\Local\Temp\VMBuilds\LEDsynthPro81-V2\esp32_esp32wrover\Release\multitasking/TaskScheduler.h(543): error 0x40081776:_ZL16loadAndTransposePh13OffsetDisplayPtiS_S_S_S_S_S_S_$isra$1
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996:::0x4008b363:xQueueGenericSend
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089ab3:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323:::0x40087945:_xt_medint3
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1):::0x40089ab0:spi_flash_op_block_func
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62:::0x40085c0f:ipc_task
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x4008b921:vPortTaskWrapper

midikidi, 2021-11-26:
not sure whats going on tried sending task schedular as zip or .h github wont let me but its in the dropbox link i sent b4

midikidi, 2021-11-26:
link to alternative multitasking instead of rtos
https://github.com/arkhipenko/TaskScheduler?fbclid=IwAR2q3WtkZOSkVGN3QtKzY3YkfrchbiTPTKhoILos-Y5nTKdVfHbl2G55XNU

midikidi, 2021-11-26:
logged the crashes 14:36:16, 14:37:41, 14:45:26, 14:47:21, 14:57:07, 15:10:16, 15:14:15

closed 2021-12-20 by hpwit