hpwit/FASTLEDVIRTUALDRIVER #11: mux 8:1 virtual driver

midikidi, 2021-12-16:
Hi Yves the reboot issue on this driver seems ok now problem was a portdelay issue in multitasking code I am using
however I have discovered another issue tested with different hardware (problem still the same)

system configuration 100 leds per strip
fill 50% strip red 50% strip white NO PROBLEM
pins defined
#define LATCH_PIN 4
#define CLOCK_PIN 21 
#define NBIS2SERIALPINS 6 
int pins[6] = { 13,14,25,26,33,32 };

change config to
#define LATCH_PIN 4
#define CLOCK_PIN 21 
#define NBIS2SERIALPINS 8 
int pins[8] = { 13,14,25,26,33,32,15,27 };
run same test code fill 50% strip red 50% strip white 
but pixel strips show 75% red 25% white

hpwit, 2021-12-17:
Hello can I see the full code please ?

midikidi, 2021-12-17:
Sent code to your gmail

hpwit, 2021-12-17:
I have seen the code can you tell me where you set up 50% red 50%white ?

midikidi, 2021-12-17:
in main ino
void LSPROfillleds(int stripnumber,int start, int end, int r, int g, int b)
{

	int s = stripnumber - 1;
	int sindex = s * NUM_LEDS_PER_STRIP;
	for (int s = start; s <end+start; s++) {
		leds[s+sindex] = CRGB(r, g, b);

		//leds[s + sindex] = r;
		//leds[s + sindex+1] = g;
		//leds[s + sindex+2] = b;
	}
}

this is called from tasks in mulitasking.h
see
void t1Callback()	{
	LSPROfillleds(1, 0, 50, 255, 0, 0);
	update = true;
}
void t2Callback() {	
	LSPROfillleds(2, 0, 50, 255, 0, 0);
	update = true;
}
howerver i discovered a typo in 

void t61Callback() {
	LSPROfillleds(1, 0, 50, 255, 0, 255);//start should be 50
	update = true;
}
void t62Callback() {
	LSPROfillleds(2, 0, 50, 255, 0, 255); //start should be 50
	update = true;
}

but still same problem i explained

midikidi, 2021-12-17:
if it helps i can remove all the multitasking stuff and put all led fills in main loop
it maybe easier for u to debug

midikidi, 2021-12-17:
simplified code with no multitasking try the 2 different pin configurations
;
// 13,14,25,26,33,32,15,27
#include <WiFi.h>
#include "FastLED.h"

#define NBIS2SERIALPINS 6 //the number of virtual pins
//#define NBIS2SERIALPINS 8 //the number of virtual pins

#define STATIC_COLOR_GRB 1 //set the strip color
#define NUM_LEDS_PER_STRIP 100
#define NUM_LEDS (NUM_LEDS_PER_STRIP*NBIS2SERIALPINS*8)
#define LATCH_PIN 4
#define CLOCK_PIN 21 
#define NUM_STRIPS 64
#include "I2SClocklessVirtualLedDriver.h"

CRGB leds[NUM_STRIPS*NUM_LEDS_PER_STRIP];
//int pins[8] = { 13,14,25,26,33,32,15,27 };
int pins[6] = { 13,14,25,26,33,32 };

I2SClocklessVirtualLedDriver driver;

void LSPROfillleds(int stripnumber,int start, int end, int r, int g, int b)
{

	int s = stripnumber - 1;
	int sindex = s * NUM_LEDS_PER_STRIP;
	for (int s = start; s <end+start; s++) {
		leds[s+sindex] = CRGB(r, g, b);
	}
}

void setup() {
	Serial.begin(230400);
	delay(1000);

	WiFi.mode(WIFI_STA); //Optional
	WiFi.begin("AirTies_Air5453", "cassie01");
	Serial.println("\nConnecting");

	while (WiFi.status() != WL_CONNECTED) {
		Serial.print(".");
		delay(100);
	}

	Serial.println("\nConnected to the WiFi network");
	Serial.print("Local ESP32 IP: ");
	Serial.println(WiFi.localIP());
	driver.initled((uint8_t*)leds, pins, CLOCK_PIN, LATCH_PIN);
}

void loop() {

		LSPROfillleds(1, 0, 10, 255, 0, 0);
		LSPROfillleds(2, 0, 10, 0, 255, 0);
		LSPROfillleds(3, 0, 10, 0, 0, 255);
		LSPROfillleds(4, 0, 10, 0, 255, 255);
		LSPROfillleds(5, 0, 10, 255,0 , 255);
		LSPROfillleds(6, 0, 10, 255, 255, 0);
		LSPROfillleds(7, 0, 10, 255, 255, 255);
		LSPROfillleds(8, 0, 10, 255, 0, 0);
		LSPROfillleds(9, 0, 10, 0, 255, 0);
		LSPROfillleds(10, 0, 10, 0, 0, 255);
		LSPROfillleds(11, 0, 10, 0, 255, 255);
		LSPROfillleds(12, 0, 10, 255, 0, 255);
		LSPROfillleds(13, 0, 10, 255, 255, 0);
		LSPROfillleds(14, 0, 10, 255, 255, 255);
		LSPROfillleds(15, 0, 10, 255, 0, 0);
		LSPROfillleds(16, 0, 10, 0, 255, 0);

		LSPROfillleds(17, 0, 10, 255, 0, 0);
		LSPROfillleds(18, 0, 10, 0, 255, 0);
		LSPROfillleds(19, 0, 10, 0, 0, 255);
		LSPROfillleds(20, 0, 10, 0, 255, 255);
		LSPROfillleds(21, 0, 10, 255, 0, 255);
		LSPROfillleds(22, 0, 10, 255, 255, 0);
		LSPROfillleds(23, 0, 10, 255, 255, 255);
		LSPROfillleds(24, 0, 10, 255, 0, 0);
		LSPROfillleds(25, 0, 10, 0, 255, 0);
		LSPROfillleds(26, 0, 10, 0, 0, 255);
		LSPROfillleds(27, 0, 10, 0, 255, 255);
		LSPROfillleds(28, 0, 10, 255, 0, 255);
		LSPROfillleds(29, 0, 10, 255, 255, 0);
		LSPROfillleds(30, 0, 10, 255, 255, 255);
		LSPROfillleds(31, 0, 10, 255, 0, 0);
		LSPROfillleds(32, 0, 10, 0, 255, 0);

		LSPROfillleds(33, 0, 10, 0, 0, 255);
		LSPROfillleds(34, 0, 10, 0, 255, 255);
		LSPROfillleds(35, 0, 10, 255, 0, 255);
		LSPROfillleds(36, 0, 10, 255, 255, 0);
		LSPROfillleds(37, 0, 10, 255, 255, 255);
		LSPROfillleds(38, 0, 10, 255, 0, 0);

		LSPROfillleds(39, 0, 10, 0, 255, 0);
		LSPROfillleds(40, 0, 10, 0, 0, 255);
		LSPROfillleds(41, 0, 10, 0, 255, 255);
		LSPROfillleds(42, 0, 10, 255, 0, 255);
		LSPROfillleds(43, 0, 10, 255, 255, 0);
		LSPROfillleds(44, 0, 10, 255, 255, 255);

		LSPROfillleds(45, 0, 10, 255, 0, 0);
		LSPROfillleds(46, 0, 10, 0, 255, 0);
		LSPROfillleds(47, 0, 10, 0, 0, 255);
		LSPROfillleds(48, 0, 10, 0, 255, 255);

		LSPROfillleds(49, 0, 10, 255, 0, 0);
		LSPROfillleds(50, 0, 10, 0, 255, 0);
		LSPROfillleds(51, 0, 10, 0, 0, 255);
		LSPROfillleds(52, 0, 10, 0, 255, 255);
		LSPROfillleds(53, 0, 10, 255, 0, 0);
		LSPROfillleds(54, 0, 10, 0, 255, 0);
		LSPROfillleds(55, 0, 10, 0, 0, 255);
		LSPROfillleds(56, 0, 10, 0, 255, 255);
		LSPROfillleds(57, 0, 10, 0, 0, 255);
		LSPROfillleds(58, 0, 10, 0, 255, 255);
		LSPROfillleds(59, 0, 10, 0, 0, 255);
		LSPROfillleds(60, 0, 10, 0, 255, 255);
		LSPROfillleds(61, 0, 10, 0, 0, 255);
		LSPROfillleds(62, 0, 10, 0, 255, 255);
		LSPROfillleds(63, 0, 10, 0, 0, 255);
		LSPROfillleds(64, 0, 10, 0, 255, 255);
		driver.showPixels(); 
}

midikidi, 2021-12-17:
Just looked in i2c driver at top of file looks like theres a define for virtual pins to 7 ??

hpwit, 2021-12-17:
Yes that is normal :)
in this code when using 8 // pins what is the behavior ?
can you modified your code to
```
void LSPROfillleds(int stripnumber,int start, int end, int r, int g, int b)
{

int s = stripnumber - 1;
int sindex = s * NUM_LEDS_PER_STRIP;
for (int s = start; s <stripnumber; s++) {
	leds[s+sindex] = CRGB(r, g, b);
}
}
```

this way normally it will have the same number of leds lit as the strip number

midikidi, 2021-12-17:
fill modified
if i use 6 pin config works as expected
if i use 8 pin config it lights more than expected number of leds i think double

hpwit, 2021-12-17:
Ok I think I know the issue I am not at home now I can give you the fix later today

hpwit, 2021-12-18:
Hello
If mal the leds are behaving strangely add this before declaring my library 
`#define NOYSCROLL 1`
Let me know

midikidi, 2021-12-18:
Ok added the #define (your modified fill func)
enabled the 8 pins now the correct number of  pixels are show. (but artifacts)
reverted to my 10 pixel fill function
new issue  on 1st strip an extra pixel    flashes(artifact), 2nd strip 2 extra pixel s   flashes(artifact)
this happens across all strips odd even

midikidi, 2021-12-18:
if i leave your #define in and use int [6] array/pins no artifacts

hpwit, 2021-12-18:
hello use this commit of the library 
https://github.com/hpwit/I2SClocklessVirtualLedDriver/tree/025add0a11d8188fe49233d505644bf088727ad7
let me know
Yves

midikidi, 2021-12-18:
replaced driver
i get compile error
I2SClocklessVirtualLedDriver.h:653: note  candidate  void I2SClocklessVirtualLedDriver  initled(uint8_t*, int*, int)
   void initled(uint8_t *leds, int *Pinsq, int CLOCK_PIN, int LATCH_PIN)
I2SClocklessVirtualLedDriver.h:653: note    candidate expects 3 arguments, 4 provided

this is my init as before
driver.initled((uint8_t*)leds, pins, CLOCK_PIN, LATCH_PIN);

hpwit, 2021-12-18:
This is strange 
But comment line 542 to 545

midikidi, 2021-12-18:
commented out but theres another initleds @ line 654

hpwit, 2021-12-18:
You still have this error code ? Normally it should compile. I am a not home right now I will later compile it

midikidi, 2021-12-18:
have fun

midikidi, 2021-12-18:
really confused both functions have same function prototypes
dont understand whats going on

#include "NEWI2SClocklessVirtualLedDriver.h"
// LINE 654 REPLACEMENT DRIVER ENRTY POINT
//void initled(uint8_t *leds, int *Pinsq, int CLOCK_PIN, int LATCH_PIN)
//------------------------------------------------------------------------

//------------------------------------------------------------------------
//#include "ORIGINALI2SClocklessVirtualLedDriver.h"
//ORIGINAL VIRUAL DRIVER ENTRY POINT
//void initled(uint8_t *leds, int *Pinsq, int clock_pin, int latch_pin)
//------------------------------------------------------------------------

hpwit, 2021-12-18:
Me too so I guess the issue is coming from somewhere else but I need a real screen to check it out. Can you  try to compile the examples of the library (the branch I gave you)

hpwit, 2021-12-18:
You only have one library present in the arduino library directory right ?

midikidi, 2021-12-18:
the 2 versions of the driver original and latest have been renamed (appended NEW and OLD they do not exist in librayrs folder they appear in the local folder with the ino file

midikidi, 2021-12-18:
branch is full f errors mismatches and witout editing ino it wont build

midikidi, 2021-12-18:
OK SO COPIED THE VIRTUAL DRIVER FROM YOUR LATEST MAIL
I2SClocklessVirtualLedDriver.h

then copied getting started from examples file
project has an include
#include "I2SClocklessLedDriver.h"

ok I can change the ino for correct header file

I can see
int pins[16] = { 0,2,4,5,12,13,14,15,16,18,19,21,22,23,25,26 };

but where is CLOCK and LATCH pins defined??

hpwit, 2021-12-18:
forget about it, it was a bad commit from my side
I will look at it now with my laptop
and see why it's not compiling

hpwit, 2021-12-19:
hello here is the file to use
[I2SClocklessVirtualLedDriver.h.zip](https://github.com/hpwit/FASTLEDVIRTUALDRIVER/files/7740145/I2SClocklessVirtualLedDriver.h.zip)
let me know

midikidi, 2021-12-19:
installed latest driver all strips addressed with correct number of filled pixels but artifacts on 2 extra leds same for 6 pins or 8 a previous version had no artifacts but double the number of fills expected Ive gmailed you a short vid of whats happening

hpwit, 2021-12-19:
hello
please try this one the
[I2SClocklessVirtualLedDriver.h.zip](https://github.com/hpwit/FASTLEDVIRTUALDRIVER/files/7740755/I2SClocklessVirtualLedDriver.h.zip)

midikidi, 2021-12-19:
Still artifacts carried out more tests:
6 pin array no artifacts
7 pin array occasional artifacts.
8 pin array artifacts appear in pixel 11,12 (10 pixel fill test)

so the additinal pins are 15,27
so to see if it was a pin issue 1 configured a 7 pin array and included pin 27 not 15
still artifacts it seems a timing issue if you go above a 6 bit array

hpwit, 2021-12-19:
How do you power the all thing ?

hpwit, 2021-12-19:
Can you try without wifi on?

midikidi, 2021-12-19:
removed wifi connect and added Wifi.mode(WIFI_OFF)
still artifacts

hpwit, 2021-12-19:
The artifacts that you see on the 7th array is that above the 10th leds ?
What happen if you stay in 8 pins but unplug the 6 first ?

hpwit, 2021-12-19:
That is strange that this issue occurs after only 8pins I will take out leds and my boards

midikidi, 2021-12-19:
I power the leds from 50 amp meanwell psu
i am only testing with 2 strips of 100 pixels moving them to different output pins
0NCE 8 pins are enabled artifacts appear on 11 and 12 pixels all strips
if your at a loss can you tell what esp32 core library you are using im using 1.0.4

hpwit, 2021-12-19:
I will Check my version.  
 Can i have to  ode of your 10 pix test ? How long is the strip ?

hpwit, 2021-12-19:
I am using version 2.0.0 on arduino

midikidi, 2021-12-19:
WS2812B 144/M AND 60/M

midikidi, 2021-12-19:
TESTING WITH 2 STRIPS 100 PIXELS EACH  144/M

midikidi, 2021-12-19:
tps://cpldcpu.wordpress.com/2014/01/14/light_ws2812-library-v2-0-part-i-understanding-the-ws2812/

midikidi, 2021-12-19:
[image]

hpwit, 2021-12-20:
> WS2812B 144/M AND 60/M

I meant how many leds in the strip 

> [image]

The issue you may experiencing comes from the speed or some interupts. Did you upgrade the core library form 1.0.4 to 2.0.0 ?

midikidi, 2021-12-20:
i usually use the latest libs in everything i do but found many issues with core 2.0.0 and thats why i reverted back to the stable 1.0..4 however i will re install  2.0 and try again

hpwit, 2021-12-20:
Oh oki I did not know about that 
But I will have a look at it. 
try the 2.0 let me know if there is an issue. But indeed I have noticed issue with the driver when using a lot of pins when changing core version.

midikidi, 2021-12-20:
LASTEST core 2.0.1 fixed all the issues.it looks like a lot was fixed since 2.0.0 . I DO GET A FLASH /artifact on the last pixel in the strip.(with or without wifi connected) As I said for the test I am using 2 lengths of 100 pixels but if I set fill to 101 (outside the physical size of the strip the full 100's are clean. THANKYOU for all your efforts. I will continue testing and keep you advised. AGAIN MANY THANKS. NOW I JUST NEED TO SEE WHAT ISSUES I HAVE WITH 2.0.1 with my other non related LED projects. Do you think the latest core will allow me to run the apa102 driver at a higher frequency

midikidi, 2021-12-20:
NOT SURE if you have time or a setup configured just curious try loading this demo 100 sketch and look at the rainbow effect it doesn't look fluid and there are momentary quick pulses on the pixels

;
// 13,14,25,26,33,32,15,27
#include <WiFi.h>
#include "FastLED.h"

#define NBIS2SERIALPINS 8 //the number of virtual pins
#define STATIC_COLOR_GRB 1 //set the strip color
#define NUM_LEDS_PER_STRIP 120
#define NUM_LEDS (NUM_LEDS_PER_STRIP*NBIS2SERIALPINS*8)
#define LATCH_PIN 4
#define CLOCK_PIN 21 
#define NUM_STRIPS 64
#define NOYSCROLL 1
#define FRAMES_PER_SECOND  120

#define FILL_SIZE  340
//------------------------------------------------------------------------
#include "I2SClocklessVirtualLedDriver.h"

CRGB leds[NUM_STRIPS*NUM_LEDS_PER_STRIP];
int pins[8] = { 13,14,25,26,33,32,15,27 };

I2SClocklessVirtualLedDriver driver;

long time1, time2, time3;

void LSPROfillleds(int stripnumber,int start, int end, int r, int g, int b)
{
	int s = stripnumber - 1;
	int sindex = s * NUM_LEDS_PER_STRIP;
	for (int s = start; s <end+start; s++) {
		leds[s+sindex] = CRGB(r, g, b);
	}
}

void LSPROfillledsyves(int stripnumber, int start, int end, int r, int g, int b)
{

	int s = stripnumber - 1;
	int sindex = s * NUM_LEDS_PER_STRIP;
	for (int s = start; s < stripnumber; s++) {
		leds[s + sindex] = CRGB(r, g, b);
	}
}

void setup() {
	Serial.begin(230400);
	delay(1000);
	//WiFi.mode(WIFI_OFF);
	WiFi.mode(WIFI_STA); 
	WiFi.begin("AirTies_Air5453", "cassie01");
	Serial.println("\nConnecting");

	while (WiFi.status() != WL_CONNECTED) {
		Serial.print(".");
		delay(100);
	}

	Serial.println("\nConnected to the WiFi network");
	Serial.print("Local ESP32 IP: ");
	Serial.println(WiFi.localIP());
	driver.initled((uint8_t*)leds, pins, CLOCK_PIN, LATCH_PIN);
}

void loop()
{
	//filltest();
	demo100test();
[... 146 lines cut ...]
	fadeToBlackBy(leds, NUM_LEDS, 20);
	int pos = beatsin16(13, 0, NUM_LEDS - 1);
	leds[pos] += CHSV(gHue, 255, 192);
}

void bpm()
{
	// colored stripes pulsing at a defined Beats-Per-Minute (BPM)
	uint8_t BeatsPerMinute = 62;
	CRGBPalette16 palette = PartyColors_p;
	uint8_t beat = beatsin8(BeatsPerMinute, 64, 255);
	for (int i = 0; i < NUM_LEDS; i++) { //9948
		leds[i] = ColorFromPalette(palette, gHue + (i * 2), beat - gHue + (i * 10));
	}
}

void juggle() {
	// eight colored dots, weaving in and out of sync with each other
	fadeToBlackBy(leds, NUM_LEDS, 20);
	byte dothue = 0;
	for (int i = 0; i < 8; i++) {
		leds[beatsin16(i + 7, 0, NUM_LEDS - 1)] |= CHSV(dothue, 200, 255);
		dothue += 32;
	}
}

midikidi, 2021-12-20:
grafted the changes and new driver into LEDSynth got 123 multitasking loops running across 18 strips no glitches

hpwit, 2021-12-20:
hello
_grafted the changes and new driver into LEDSynth got 123 multitasking loops running across 18 strips no glitches_
Happy to hear that, do not hesitate to put a video out there

I think the 
```
driver.showPixels();
time2 = ESP.getCycleCount();
driver.showPixels();
```
is not too much of the taste of the driver because the two showPixels() are too close to eah other i will need to increase the last temp buffer 

_I DO GET A FLASH /artifact on the last pixel in the strip.(with or without wifi connected) As I said for the test I am using 2 lengths of 100 pixels but if I set fill to 101_
I need to have a look at that because the driver was developped with an older version of the core and I don"t always retest everything with new core release (which I should do) and i remember it was a mess to set up the final interupt ...

so it means above 6/7 // pins the capability of doing hardware scrolling is a bit not possible
Thank you for testing it.
THANK YOU FOR USING MY LIBRARY I am happy to see it used

midikidi, 2021-12-20:
Thankyou for your brilliant work I am always pleased to work with you and kick the arse out of your code
apart from about 15 other projects Im working on a pc app timeline player that allows you to put osc, artnet, midi and dmx events onto a timeline, if I find time Iam planning to put a video together of a live midi controlled sequence like one of the early videos i posted found a brillaint peice of music  (copyrighted) but I will gmail when done. Similar to your fulltime job  I have a kinda daytime job (self employed) writing software, feeds wife and daughter but developing my own brand of pixel controller LEDSYNTH im GONNA GMAIL U MY DETAILS WOULD LOVE TO WHATTSAPP

hpwit, 2021-12-21:
Hello
you are really busy !!!
I have mailed you my contacts

closed 2021-12-27 by midikidi